cleanflight/cleanflight #1415: Horizon or level changes during fast flying

l2smith2, 2015-10-20:
I have found AFTER fast forward flying either in acro or horizon mode that what was originally calibrated as level is no longer correct.
The quad will drift/lean considerably in what should be level stable hover. 

The last few versions of cleanflight have similar results.
I have seen it reported in BETAFLIGHT as well.

Power cycling fixes the problem.
Sometimes holding a hover for a few seconds it will drift back to "level"
Sometime arming and disarming will also reset "level"

digitalentity, 2015-10-20:
I can confirm this issue with current IMU. Estimated horizon level drifts when quad is experiencing fast acceleration or deceleration. I have posted a PR #1411 that considerably improves IMU accuracy and reduces effects of level drift.

borisbstyle, 2015-10-20:
@digitalentity Would it be nice if you can provide him a binary to test and verify this. Or perhaps doing a PR in betaflight to get more users to test this?

digitalentity, 2015-10-20:
@l2smith2 
You can grab a binary with new IMU here: https://www.dropbox.com/sh/4r5374mos7471q2/AAD4ah-K7vfA5cIDUgSXOtAba/PR1411?dl=0
Give it a try :smile:

@borisbstyle Good idea. I'll try to put something together and post a PR to betaflight.

lostcontrol, 2015-12-07:
Since I'm a terrible pilot (well I'm learning) I mainly flight in ANGLE mode and also noticed this problem. I have built and flashed the latest #1411 and I hope to test it soon. Hopefully it will solve/improve the attitude estimate problem.

@digitalentity I have seen https://github.com/borisbstyle/betaflight/commit/621008cffc721e98779197fd152f5763bf904a63 which is related to this change. Should I cherry-pick this one too?

digitalentity, 2015-12-07:
@lostcontrol #1411 is enough to try out the new code. Already includes all the IMU-related changes from betaflight.

lostcontrol, 2015-12-07:
@digitalentity Fine. Thank you. Will see if I find time this week to test it a bit.

hydra, 2015-12-13:
@lostcontrol Did you get a chance to test #1411 ?

lostcontrol, 2015-12-14:
@hydra Unfortunately not yet... Short daylight (almost dark when I come back home after work) and a very busy week-end kept me away from my quad :cry: The hex is actually ready on my quad but I don't flight anymore in a rush. I had a few bad experiences wanted to quickly try something out :wink: I saw you merged the quaternion change so I will compile master and give it a try whenever I find some spare time.

lostcontrol, 2015-12-15:
I quickly flight with master as of 2150904 this evening (well... 5pm). It was getting dark so I didn't flight long but it seems much better now :smile: Thank you.

hydra, 2016-03-01:
ok, closing.  there seem to be some other issues related to this topic so we can discuss elsewhere if needed.

closed 2016-03-01 by hydra