cleanflight/cleanflight #1417: F3 Targets Not supporting EMF avoidance

borisbstyle, 2015-10-20:
I was wondering what the reason is that F3 targets have unchangable system clock like F1 boards?
Just not implemented or something else?

EMF avoidance is still very usefull with UHF systems.

readerror67, 2015-10-20:
The function for setting the clock doesn't seem to exist for the F3 in CMSIS, here is where its done for the F1:
https://github.com/cleanflight/cleanflight/blob/master/lib/main/CMSIS/CM3/DeviceSupport/ST/STM32F10x/system_stm32f10x.c#L90

borisbstyle, 2015-10-20:
@readerror67 I saw that, but I guess with some effort the function for F3 could also be integrated?

readerror67, 2015-10-20:
I suppose it could, for some reason the system_stm32f30x.c file for each F3 target is in the same directory as target.h, unsure the reason for this since they all seem to be identical..

Would make sense to move this over to the CMSIS path and then get the changes made for overclocking.

readerror67, 2015-10-20:
https://github.com/multiwii/baseflight/commit/6716ecfe7e20d9c4922a233576d391762d5cbdff?diff=split

The change would be even easier since there are no F3 targets using the 12mhz external osc, its just a matter of changing PPL MUL to 10 rather than 9.

borisbstyle, 2015-10-20:
I see. Looks easy as expected. But first all f3 system files would need a cleanup I would say.

@hydra What do you think about this?

hydra, 2015-10-20:
Well, from a vendors perspective overclocking anything is not a good idea.
1) more heat.
2) more power.
3) shorter product life
4) annoyed users (if a board fails)
5) annoyed retailers (they cant prove failure reason on a dead cpu)
6) annoyed manufacturers

Yes it may be useful but a filter or RF shield is safer all round.

hydra, 2015-10-20:
Certainly, I would want solid hard evidence that there is zero risk in allowing users to change it.
I2C overclock PR has the same issues.

digitalentity, 2015-10-20:
@hydra, probably a better suggestion is to underclock the CPU to 64 mhz.
This will move harmonics away from 433 mhz band and won't break USB as well.

digitalentity, 2015-10-20:
My bad, USB will still malfunction, but issues you raised would be resolved.

borisbstyle, 2015-10-20:
@hydra I partly agree with you. This would right argumentation if there wasnt a EMF avoidance on F1 targets. Now UHF users who wants to upgrade to F3 boards can't do that easily as it misses a key feature they need. There should be more consistent feature support between the targets.
Either it should be removed for F1 targets or added to F3 targets.

borisbstyle, 2015-10-21:
closing as it is clear that it will not be there

closed 2015-10-21 by borisbstyle

digitalentity, 2015-10-21:
This is bad. It means I'd be unable to use 433MHz LRS with my SPRF3 :(

borisbstyle, 2015-10-21:
@digitalentity Obviously. Same issue here. Will need to shield it perhaps, but not much room on my mini :(

digitalentity, 2015-10-21:
Shielding won't help much, unless you'll have a ferrite ring on every single wire going to the FC and the FC itself in the solid metal case.