cleanflight/cleanflight #1428: Cleanflight/Naze32 Control channels not responding

pilotgeek, 2015-10-24:
So Im having an issue where the throttle, rudder, ail, elev channels dont work anymore, but the Aux switches do. I've tried everything to figure this out including, switching rx's, triple checking the wire harness from the naze to the rx, erasing the Naze32 board and re-loading the firmware, calibrating the accelerometer etc. I'm all out of ideas. Can someone please help sort this out?

Here's a video of whats happening for clarity:
https://youtu.be/hlkhvaQFAro

I just did a continuity test with my multimeter from the solder on the pins to the end of the signal wire. It's definitely not a connection problem.

What else would cause the stick controls to not register any movements? If I move one of the aux signal wires to the throttle slot on the rx, the throttle stick slides the aux1 in cleanflight receiver tab, so its not the transmitter. That only leaves the rx or the naze32. I've used 3 rx's with the same result, so that leaves the naze. Is there anything that would cause a lockout of the control surfaces? Throttle, Ail, Elev, Rudder?

SteveAmor, 2015-10-24:
Sorry for your problem but this is a support question, not an issue. Please consider asking such questions in a support environment such as RC Groups or #cleanflight, rather than diverting developer time. 
Please can you close this issue. Thank you.

closed 2015-10-24 by pilotgeek

SteveAmor, 2015-10-24:
I hope it fixed your problem.  Thanks for closing the issue.

pilotgeek, 2015-10-24:
Hi,

It didn't really fix it.  I think this may actually be more of a bug than a support issue.  Chrome released an update 2 days ago to version 46.0.2490.80 m  Since then I've been having all these issues.  On the configurator tab, the save and reboot doesnt re-boot anymore.  Also, I flashed back to Naze 1.9.0 and now all channels respond except elevator.  This is with a new/clean model created in my tx.  Tx monitor shows outputs are fine and I confirmed that by switching 2 signal wires.  I believe this to be a bug of some sort as everything worked as designed last week using Naze 1.10.0 firmware.

SteveAmor, 2015-10-24:
I you really see a bug then please raise an issue under cleanflight/configurator