cleanflight/cleanflight #1432: Profiles 1&2 save duplicate settings

DaDot, 2015-10-25:
When using same PID type (Luxfloat in my case) using profiles 1 & 2, last rate settings are overwritten on the opposing profile.
i've confirmed this isnt the case with profile 3. i.e. when using luxfloat roll rate, pitch rate, yaw rate, rc rate... etc are saved properly regardless of profile 1 and 2 settings.
I suspect it's a 0,1 indexing error in the code.  It's persisted over many versions. 
Surprised noone else has caught this.

stronnag, 2015-10-25:
Works correctly for me across multiple FCs and versions.
Sure you've never copy & pasted dump data in the CLI, and thus copied 
````
set default_rate_profile = 0
````
from profile 0 to profile 1, so they both refer to the same rateprofile. Normally, for profile 1:
````
set default_rate_profile = 1
````
Should this be the case, fix it in the CLI (unique default_rate_profile per profile), verify in the GUI and close the issue if fixed.

DaDot, 2015-10-26:
i've dumped all 3 profiles and they look like this:
(default_rate_profile = rateprofile = [pid profile i selected in gui]-1)

....
set default_rate_profile = 2
set gimbal_mode = 0
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff =  5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf =  0.600
set baro_cf_vel =  0.985
set baro_cf_alt =  0.965
set mag_declination = 0
set pid_controller = 2
set p_pitch = 30
set i_pitch = 30
set d_pitch = 23
set p_roll = 30
set i_roll = 30
set d_roll = 23
set p_yaw = 75
set i_yaw = 45
set d_yaw = 0
set p_pitchf =  3.000
set i_pitchf =  0.450
set d_pitchf =  0.018
set p_rollf =  2.400
set i_rollf =  0.390
set d_rollf =  0.015
set p_yawf =  9.500
set i_yawf =  0.450
set d_yawf =  0.003
set level_horizon =  1.000
set level_angle =  5.000
set sensitivity_horizon = 100
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 0
set i_vel = 0
set d_vel = 0
set yaw_p_limit = 500
set dterm_cut_hz = 0
set pterm_cut_hz = 0
set gyro_cut_hz = 0

# dump rates

# rateprofile
rateprofile 2

set rc_rate = 70
set rc_expo = 0
set rc_yaw_expo = 0
set thr_mid = 50
set thr_expo = 0
set roll_rate = 23
set pitch_rate = 27
set yaw_rate = 24
set tpa_rate = 20
set tpa_breakpoint = 1500

closed 2017-02-25 by hydra