cleanflight/cleanflight #1433: Waiting for data after reboot on changing tabs

ProDrone, 2015-10-25:
It is impossible to change tabs after a reboot via the configurator tab PORTS and a click on Save and Reboot. The tab window stays blank with the message waiting for data.

After a Disconnect / reconnect all is normal again.

Initially if made an issue for configurator (https://github.com/cleanflight/cleanflight-configurator/issues/261), but it turns out this is a combination of boot behavior and changes in the chrome serial implementation. See https://github.com/cleanflight/cleanflight-configurator/issues/261#issuecomment-150839710

__*Almost solved*__ :smile:

Trussroads, 2015-10-25:
> Initially if made an issue for configurator (cleanflight/cleanflight-configurator#261), but it turns out this is a combination of boot behavior and changes in the chrome serial implementation. See cleanflight/cleanflight-configurator#261 (comment)

Ive been getting a lot of this too.. First I thought it was to do with latest FW version, but it does not occur on the F3 boards I have and appears limited to the F1's? Both acro and full version behave as described below.
It does hold settings and there is no other effects noticed on mine, but the usb/board must be cycled to re-activate the GUI. If it is GUI/Chrome issue, that would fit with what I am seeing.

seattleite09, 2015-10-26:
I have seen this as well, it started recently. I initially thought my Naze32 was busted (after a crash), but that doesn't seem to be the case. All of my Naze32/F1 boards behave this way.

Rhyell, 2015-10-26:
Try using an older version of chrome as it looks as though this happened just after a chrome update.

ProDrone, 2015-10-26:
@Rhyell Thanks, but i prefer to go forward (going back would prevent chrome from auto-updating security fixes).

The problem should be fixed with pull request https://github.com/cleanflight/cleanflight/pull/1436

ProDrone, 2015-10-30:
Solved with #1436

closed 2015-10-30 by ProDrone