cleanflight/cleanflight #1435: CF 1.9 issue with auto-disarming in flight

seattleite09, 2015-10-25:
I am not (yet) convinced this issue applies in CF 1.10, someone with more experience with the code should confirm or shoot this down, as appropriate.

In looking at the CF 1.9 code, the overall behavioral issue is:
- say rcData[THROTTLE] < min_throttle, even for a single loop iteration, e.g. as a result of an RX glitch or noise.
- in mw.c, processRx():
- processRx calls calculateRxChannelsAndUpdateFailsafe, line 530, which increments the failsafe counter via failsafeOnRxCycleStarted, still under the threshold though (!).
- processRx then calls on failsafeUpdateState, line 546
- failsafeUpdateState senses that the counter < threshold, and maintains its state as IDLE, and active=false. Nothing else is done in failsafe.c at this iteration (!).
- Back in processRx(), the throttle_status will be determined as THROTTLE_LOW
- processRcStickPositions is called, which checks the failsafe status (rc_controls.c, line 129), and since it's not active, it just calls mwDisarm().
- if disarm_kill_switch happens to be 1, it doesn't even matter that the throttle is low. The board seems to be disarmed anyway.
==> quad will drop from the sky(?).

The premise here is:
- disarm_kill_switch=0 or 1, doesn't matter
- arming via a switch (not using sticks to arm)
- FC failsafe enabled

Am I misreading the code? The real problem seems to be in failsafe's management of its counter, and the fact that it doesn't acknowledge the RX glitch immediately (stays active=false, state=IDLE), effectively letting the subsequent stages in the loop iteration deal with the bad data. This doesn't seem safe, and can result in a mwDisarm mid-flight.

I see the failsafe code has changed considerably in 1.10, but I have not yet fully internalize it, so I can't say if the problem is also present in 1.10. Anyone?

seattleite09, 2015-11-16:
Disabling MOTOR_STOP (feature -motor_stop) and setting disarm_kill_switch=0, with FC failsafe enabled @ failsafe_throttle == min_throttle (spinning) seems to prevent automatic disarms. I haven't had one since the previous post. So this seems to be a problem with this particular combination of settings, as I suspected when I read the code 21 days ago... This is a bug that really should be fixed.

hydra, 2016-01-11:
@ProDrone can you look into this please

hydra, 2016-01-21:
@seattleite09 Please see #1710 for a different approach.  I would love it if you could do some testing of the new PR and give feedback on it.

closed 2017-02-25 by hydra