cleanflight/cleanflight #1437: Headfree mode suggestion

arrowcircle, 2015-10-26:
Hi! 
Headfree mode now works strange. It stores direction only when activated headfree mode.
Is it possible to make it lock direction when FC is armed?

OleEngr, 2015-11-08:
I'll second the above!  I'm surprised and shocked to learn it works as reported above.

Without the capability to lock onto a known bearing when it is ground,  it is worthless as a pseudo return to home.  Without that, I'm not sure what the worth of headless would be!

I've seen too many videos by experts that put down headless while seeming to have no clue of the capability the mag compass brings in headless mode when it is referenced to a heading you know.  When the initial heading is established at boot up or arming, it gives you the capability to fly it back to your location.  I've lost quads (due to my own errors) when they got out of my visual range and all of my attempts to recover them only flew them farther away.  

On the other hand, I've had quads with Headless that escaped my orientation ability and all I had to do was flip a switch to turn headless and then I could fly them back using a stick bound to the compass, not one bound to an unknown vehicle orientation.

robustini, 2015-11-12:
+1
The "headfree" function it's only really useful if the position of the virtual heading is "the takeoff nose direction", not when the function is activated.

joshuabardwell, 2015-11-12:
>In order to store a new virtual direction would be necessary another "AUX" for this function, then it would make sense.

FWIW, this is/was how MultiWii worked. There are/were two modes, HEADFREE and HEADADJ. HEADADJ was used to set the orientation for HEADFREE mode.

arrowcircle, 2015-11-12:
> FWIW, this is/was how MultiWii worked. There are/were two modes, HEADFREE and HEADADJ. HEADADJ was used to set the orientation for HEADFREE mode.

Thats what in the docs, but its not usable.

ChrOnOs83, 2016-02-01:
In the docs, it's wrote : 
"In this mode, the "head" of the multicopter is always pointing to the same direction as when the feature was activated."

But I have read nw.c and the only place where the variable headFreeModeHold is set is when arm. So, maybe the docs isn't correct. I will test that soon.

ChrOnOs83, 2016-02-01:
Not the only place, also when HeadFreeAdjust. But headFreeModeHold seems not reset when the headfree is activated

antwin, 2017-01-11:
I strongly support OleEngr's comment above. Is anything going to happen about this?

closed 2017-02-25 by hydra