cleanflight/cleanflight #1438: Windows10 issue ?

Fvk1960, 2015-10-26:
I lately use Windows10 operating system.
Now the sensor tab does not display any data trends and application gets stuck . Cannot change tab any more.
Need to disconnect and reconnect.
Has anyone same issue ?

joshuabardwell, 2015-10-26:
https://github.com/cleanflight/cleanflight/pull/1436#issuecomment-151240388

seattleite09, 2015-10-27:
I don't think it's a Win10 problem. It seems to be an issue with the latest Chrome runtime. As Joshua pointed out, this is being addressed.

1477, 2015-10-27:
I figured I had problems with the software but I did not know where to look.  I seems others have similar problems.  Hopefully, this issue can be addressed and we can move on with our projects

borisbstyle, 2015-10-30:
it is addressed.....just download manually the development version of configurator

Skidaddle23, 2016-09-27:
I'm tagging on to the end of this with a similar issue. Has anyone else had problems since the Aniversary Update of Windows 10 a couple of days ago? Cleanflight is laggy and it claims the battery is 1.4V without a battery plugged in and the BlHeli passthrough no longer works.

closed 2017-02-25 by hydra