cleanflight/cleanflight #1439: Serial Passthrough

mcowger, 2015-10-26:
I'd love to see a serial 'passthrough' function so I can issue commands to a serial device (like an HC05/06 Bluetooth Module)

enly1, 2015-10-26:
You suggested issuing commands - did you mean HC05 commands or simply data
strings to the module in user mode ?
Thinking that to issue 'Commands' to the HC05 it would need to be in
command mode ( i.e. KEY pins modified and resets etc ).

mcowger, 2015-10-26:
I think both.  My specific use case is for the HC06 module, which doesn't actually require a KEY pin (although I believe the 05 does).

Let me give you something more concrete (user story style):

As a *user*, I want to *send arbitrary strings to attached serial devices* so that *devices can be controlled/managed over serial if such capability is supported by the device*.

Even more concrete:

My HC06 is set for 57600, but I'd really prefer it at 115200. Its buried deep in my quad, so pulling it out, attaching it to an FTDI (which I need to go find :-1: ) just to issue a single `AT +BAUD8` command is sub optimal.  I'd like to have a cli command like `raw_serial 1` which drops me into a direct connection to that UART, along with an 'escape' sequence to exit from such a mode (even just requiring a reboot to exit would be fine).

Being able to pull specific pins high or low would be a secondary request.

tracernz, 2015-10-27:
Pretty clear what he means, just serial passthrough. So a serial connection to the board on your PC is directly passed through to one of the onboard serial ports. This is implemented in OP/TL/LP as CommBridge, so the VCP connection can be used for CommBridge while the HID connection can still be used for connection to GCS (the equivalent of configurator). This would be more difficult/less friendly on most of the CF boards because they don't have an actual USB connection so you can't create multiple endpoints.

enly1, 2015-10-27:
Thanks for the clarification - new to flight controllers, so not aware of
other implementations.  Not quite as easy on many CF boards as you say.

ledvinap, 2015-10-27:
It's not htat hard: https://github.com/ledvinap/cleanflight/blob/local/src/main/io/serial_cli.c#L1609

I just modified/replaced gpspassthrough. GPS related stuff is removed (GPS no longer tries to parse data during passthrough) and code for port,speed and port mode is added.
Return is possible only using reset, but some escape (+++) may be added quite easily.

dkisselev, 2015-10-29:
@ledvinap are you planning on submitting a PR for that passthrough change? I'm very interested, and as someone mentioned in the serial 1wire thread, it would make configuring an OSD so much easier.

nathantsoi, 2015-10-29:
this is a great idea, lmk if i can help @ledvinap

btimby, 2016-01-15:
I published a release including this feature for testing:

https://github.com/btimby/cleanflight/releases/tag/v.11.0-SerialPassthrough

hydra, 2016-11-01:
implemented, closing.

closed 2016-11-01 by hydra