cleanflight/cleanflight #1442: CC3D - unable to connect after using CLI

akcom, 2015-10-27:
To recreate:
1.  Flash CC3D with cleanflight w/ the OP bootloader.
2.  Open Cleanflight configurator
3.  Go to the CLI tab, type "save" and press enter
4.  At this point, you should no longer be able to reconnect to the board via USB or FTDI w/ mainport

Only solution is to restart cleanflight.

Tested on: OSX 10.10.5 using both USB and FTDI adapter.

joshuabardwell, 2015-10-27:
Is this the same issue?

https://github.com/cleanflight/cleanflight/pull/1436

akcom, 2015-10-28:
I do not believe so as the board does not require a reboot cycle.

IronMessenger, 2015-11-03:
I am pretty sure this is the same issue a buddy and I are seeing with Cleanflight.  We both Have Nave32 boards, and both boards and PC's have the same issue.

1. Connect to board
2. Make a change... CLI or otherwise.
3. Save - Causes reboot
4. Interface says 'rebooting' forever... 
5. Must disconnect in order to do anything further.
6. Once disconnect, and then reconnect it will work again...

seattleite09, 2015-11-11:
Was this issue fixed? Any idea when it will make it into a released (non-dev) build?

IronMessenger, 2015-11-11:
It is still an issue with my board.  I reflashed, updated again... still
same issue.

Someone told me that it is related to an issue with the newest version of
chrome.  Apparently cleanflight does something wrong and in the past the
command was ignored by chrome.  The new version of chrome does not ignore
this issue and it prevents the refresh/boot.

Aaron Parker (Mobile)

tracernz, 2015-11-11:
>Someone told me that it is related to an issue with the newest version of
chrome.  Apparently cleanflight does something wrong and in the past the
command was ignored by chrome.  The new version of chrome does not ignore
this issue and it prevents the refresh/boot.

If that's referring to the "break" condition which can sometimes occur on the Naze and other CP2102 boards, it's not applicable to the CC3D, it has a real USB. There are issues with the way CF handles the USB when combined with the OP bootloader.

gc9n, 2015-12-01:
Hi.
Do you use an OSD , i have exactly the same problem , and i came to the conclusion that OSD and USB can work together . i will test tonight to disconnect the osd

"But be aware that the computer connection uses the same serial port, that means if you want to use Baseflight, or Cleanflight configurator, you will have to disconnect the TX and RX from the OSD first."\
http://blog.oscarliang.net/minimosd-micro-setup-naze32-pid-rssi/

akcom, 2015-12-07:
gc9n: I do not use an OSD.  Regardless, that is not an issue with the CC3D which uses a VCP port for the USB plug.

closed 2017-02-25 by hydra