cleanflight/cleanflight #1457: Servo settings saved across rate profiles

CerealKiller1, 2015-10-31:
Hello all, this is my first ever issue tracker...

I've recently upgraded all my tricopters to CF1.10 and I experienced a crazy issue it took me a bit to find. One of them is using a cheap servo that operates backwards of all my others, it needed to be reversed.

When I switched profiles to try out some quicker rates I crashed instantly. After some searching I realized servo reversing settings weren't applied to all three profiles and I had to manually do that.
Would it be possible to change it so a servo reversal is automatically applied across the board to all three profiles? 
Is there a reason it doesnt work like that now? I understand wanting different servo %'s in different rates but in my opinion the user would always want the same direction setting.

Thank you all so much for everything! I'm not complaining and I did get it figured out after some tinkering but this would have kept me in the air instead of having to pack up and head home early cause I thought something was wrong.
Appreciate everything everyone here does!
-Cereal_Killer

CerealKiller1, 2015-10-31:
I guess I should add all my info...
I am running the RCExplorer integrated tricopter bodies (on three different copters, only one needed the servo reversed)
I am downloading FW from the pull down by selecting Naze32 stable, 
My exact FW revision is: 
# Cleanflight/NAZE 1.10.0 Oct  2 2015 / 14:57:31 (9f95334)

Thanks again!
-C_K

stronnag, 2015-10-31:
This was discussed on IRC during the migration to ```smix``` and as a result it is clearly documented in https://github.com/cleanflight/cleanflight/blob/master/docs/Mixer.md that ```smix``` commands are per profile. The discussion also determined that It is not practical to make this a system property.

This issue thus describes the correct behaviour of a documented feature and might be closed, please.

closed 2015-10-31 by CerealKiller1