cleanflight/cleanflight #1462: Feature request: Alternative output pin for RGB LEDs?

martgadget, 2015-11-01:
Many boards have other output pins that could be used for addressable LED strips - e.g NAZE32 has FRSky telemetry pin.

Can you consider making it possible to configure RGB led support on to say the telemetry pin instead of requiring PPM mode for example? or is this feature dependent on PPM mode for other reasons?

Thanks again for such a stable, easy to use platform - you have no idea how many happy boys enjoy their toys because of the work you folks have done 8-) - Thanks.

DaEmpty, 2015-11-24:
Being able to use one softserial port and the led strip at the same time, would be amazing. I'd like to use SBus (UART), USB (UART), MinimOSD (Softserial) and a led strip. The naze has still a lot of unused pins but the combination is not possible with cleanflight atm.

Lanzi2307, 2015-12-21:
Or maybe it's possible to use one of the motoroutputs. GPIO Pin is defined in the target.h file of the target. An idea would be to let the user select a port in gui which should be used.

ledvinap, 2015-12-21:
It's not simple. RGB output needs exclusive use of whole timer and associated DMA channel. It's quite hard to make this generic.

Lanzi2307, 2015-12-21:
Okay, thank you very much for clarifying this. I haven't seen that till now. Maybe something comes in my mind while studying the Cleanflight Code further.

closed 2017-02-25 by hydra