cleanflight/cleanflight #1464: Feature request:

croccodillo, 2015-11-03:
Hello, I have a request:
I put togheter an homebrew flight controller, using a minimal STM32F103C8 board from ebay and a chinese gy-86 IMU.
Everything seems to work fine, but the IMU seems to send out strange data.
When I flip the board, magnetometer directions goes wrong, and if I leave the board rest on the table, the direction is 180° flipped.
I found out that the magnetometer is phisically turned 180° on the PCB, so I set (via CLI) the align_mag variable to 3, but nothing has changed.
After checking the gy-86 schematic, I found also out that the magnetometer is connected to the auxiliary I2C of MPU6050, so probably this is the problem.
Would it be possible to configure the MPU6050 chip in manner to enable the I2C passthrough?
This way, it should be possible to read the data from magnetometer directly, without interpretation from MPU6050.
Any other idea?

Thank you in advance.

Giovanni Albergoni

ledvinap, 2015-11-03:
I2C passthrough should be enabled: https://github.com/cleanflight/cleanflight/blob/526f4777a24e6963ded6e6b92060350aa0e95878/src/main/drivers/accgyro_mpu6050.c#L118

croccodillo, 2015-11-03:
Oh well,

This is a bad news for me...
At this point most probably my IMU board is defective.
MPU6050 seems to work correctly, bu tnot so the magnetometer.
When phisically turned to south (X->), it indicate North, and viceversa.
if I try to flip the board 45* on side (Y+/Y-), the magnetometer goes crazy, and drifts 180° in every direction.
So, with the board sticked in the South direction, but twisted in the Y+ axis, the magnetometer drifts from south to north and back when put again flat.
There's no way to change the behaviour, even changing the CLI variable set.
Looking at the sensors graph, however, I can see that the magnetometer readings (Ga) never change, and remain always close to zero.

OK, thank you very much, I will investigate on my side.

SteveAmor, 2015-11-04:
@croccodillo have you checked the calibration and orientation of your accelerometer?  When upright and level the z axis should be 1g.

croccodillo, 2015-11-04:
Hello, yes I did.
You can see here the FC laying flat on the table:
[image]

And this is the Seosnros page in such a condition:
[image]

As you can see, X axis of the accelerometer is 1g.

If I simply "roll" the board on the right (Y+) of about 16°, this is what I end up with:

croccodillo, 2015-11-04:
Board "rolled" 16° on the right, first page:
[image]

And Se
[image]
nsors page:

croccodillo, 2015-11-04:
As you can seee, without channging the heading of the board (Z), it changes from 302° to 63°...
The gyroscope seems to work fine, it indicates the right angle, but the magnetometer is waayyy out.
But I'm afraid I have a problem with my board.

ledvinap, 2015-11-04:
Did you calibrate the magnetometer?

croccodillo, 2015-11-04:
Yes, I tried to calibrate the magnetometer several times...

ledvinap, 2015-11-05:
Strange ... how does MAG sensor trace look when rotating quad? Can you try aligning quad so that only one magnetometer axis reading is nonzero, for each axis, and post results?

closed 2017-02-25 by hydra