cleanflight/cleanflight #1467: IMU level errors after agressive rate-mode flight

digitalentity, 2015-11-05:
I've been reported several cases or machines crashing after switching to ANGLE more from rate mode. Some said the machines "forgot" about what level is. I believe I've found the source of the problem.

Gyro on our flight controller is hard limited to 2000 degress per second. However, RC rates might be configured to request rotation rate of more than 2000 dps and PID controller will try to push the actual rotation rate above this hardcoded limit.

IMU works by integrating data from gyro and slowly correcting it using accelerometer.

Imagine the quad rotating at 3600 degrees per second for 200 ms (doung a 720 deg rotation - double flip) starting from level and ending up level. That's what actually happend.
And this is what IMU calculated: gyro is reading 2000 dps for 200ms, IMU would calculate rotation or only 400 degrees and calculated attitude will be 40 degrees off.

lostcontrol, 2015-11-09:
@digitalentity Just for my information, could you quickly explain why you closed this issue? Thank you.

digitalentity, 2015-11-09:
@lostcontrol 
I figured that the issue is caused not by gyro saturation and was able to solve most cases of level errors or at least make them much less evident. PR #1411 has the related changes.

ledvinap, 2015-11-09:
@digitalentity : Can you briefly explain the changes? It is not apparent when all commits are squashed ...

lostcontrol, 2015-11-09:
@ledvinap Probably no something that should be done here but in PR #1411

ledvinap, 2015-11-09:
@lostcontrol : Gyro overflow is not IMU specific. And there are some reports related to CF IMU, it may be related

joshuabardwell, 2015-11-09:
With a cap of 2000 dps, it's hard to imagine that the gyro is ever saturating. That's 5.5. flips per second.

ledvinap, 2015-11-09:
@joshuabardwell: 5 flips/s here: https://www.youtube.com/watch?v=wk0QbFjuz00
10 seems possible (but I cant find video now)

joshuabardwell, 2015-11-09:
I can honestly say I've never seen anyone flip that fast before. Are the people who are experiencing autolevel drift on Cleanflight running rates that high?

ledvinap, 2015-11-09:
From my point of view, 2000deg/s problem is interesting when using level mode as failsafe (or auto-failsafe). It is reasonable to depend on switching to level mode to work no matter what ...

joshuabardwell, 2015-11-09:
I agree. I just don't think we're solving the problem people are actually currently having, if we're focused on saturation.

digitalentity, 2015-11-09:
I've renamed and re-opened this issue. Gyro saturation issue is not solved.

hydra, 2016-03-01:
@digitalentity what's next to do with this issue?

closed 2017-02-25 by hydra