cleanflight/cleanflight #1468: No Accel/Gyro function on Naze32 Rev6 with CleanFlight 1.10.0

CommanderBatFink, 2015-11-05:
Just bought an AfroFlight Naze32 Rev6 (which I'm led to believe is an original version) .   Flashed with CleanFlight 1.10.0.NAZE 2015-10-3 14:1 (stable).  However the firmware does not appear to be reading the sensor data from the gyro and accelerometer.   This is confirmed by no reflected movement of the simulated model when the Naze32 is moved. The GYRO and ACCEL indicators at the top right of the configurator are RED.  The MAG and BARO indicators are green.  There is an LED onboard the NAZE32 which is flashing at approximately 1sec intervals.
I have switched to BaseFlight Stable 2015.08.27-13:10.  Indicators, GYRO, ACCEL, MAG and BARO are all green, and the board appears to be functioning with the BaseFlight firmware.
Is there a problem with the firmware or am I missing some elementary set-up?

joshuabardwell, 2015-11-05:
Naze32 Rev6 is not currently supported and has been added as a 1.11 milestone. A search of the repo for "Naze32 rev6" turns up this:

https://github.com/cleanflight/cleanflight/issues/1321

CommanderBatFink, 2015-11-05:
Thank you Joshua.   I will keep my eye out for updates.
Kind regards.

closed 2015-11-05 by CommanderBatFink

SteveAmor, 2015-11-14:
Should be fixed in latest builds:

http://cleanflight.memoryleaks.org/builds/

skaman82, 2015-11-14:
Fixed in PR #275

SteveAmor, 2015-11-14:
@skaman82 please can you confirm your referenced PR.

skaman82, 2015-11-14:
@SteveAmor sorry, wrong PR. 
https://github.com/cleanflight/cleanflight-configurator/pull/275

marathonboy26, 2015-11-25:
Just received a Naze32 full rev6. Loaded CF 1.10 stable. Initially no accell or gyro. Loaded 1.10 build 2434 from above link, and accell now works. gyro works... but not until board has been powered for a few minutes. For first few minutes, gyro flat line on Sensors page of CF. After waiting... all sensors seem to work normally. Also, have needed to bridge boot loader pads each time I've flashed. Thanks guys for all your hard work!

Runfromrobots, 2015-12-23:
I'm having the same issue with a Rev5 board - gyro/accel won't register no matter what build I flash it with. Board came preflashed with cleanflight, but I've migrated back and forth to baseflight several times now trying to find a firmware that will recognize the gyro/accel...