cleanflight/cleanflight #1469: Board CC3D setting tricopter.

a6j, 2015-11-05:
Help to understand please. Board CC3D cleanflight 1.10.0 firmware installed on tricopter receiver connected by RX_PPM / RX_SERIAL.
To connect the tail servo 6 pin RC_Output?
Pin Function
1 MOTOR 1
2 MOTOR 2
3 MOTOR 3
4 MOTOR 4
5 LED Strip
6 Unused
I plugged in the socket 4, but the servo is not working and really hot.

SteveAmor, 2015-11-05:
Sorry for your problem but this is a support question, not an issue. Please consider asking such questions in a support environment such as RC Groups or #cleanflight, rather than diverting developer time and close this issue.

a6j, 2015-11-05:
Problem solved: https://github.com/cleanflight/cleanflight/issues/586#issuecomment-154161526

closed 2015-11-05 by a6j