cleanflight/cleanflight #1471: 303 I2C config

ledvinap, 2015-11-05:
I2C pins are setup as PushPull (GPIO_OType_PP) https://github.com/cleanflight/cleanflight/blob/master/src/main/drivers/bus_i2c_stm32f30x.c#L104

Is this this correct?

atomiclama, 2015-11-06:
Don't think so, it's been awhile since I setup the gpio for my day job but we used 

        /*!< Configure sEE_I2C pins: SCL */
            GPIO_InitStructure.GPIO_Pin = I2C1_SCL_PIN;
            GPIO_InitStructure.GPIO_Mode = GPIO_Mode_AF;
            GPIO_InitStructure.GPIO_Speed = GPIO_Speed_50MHz;
            GPIO_InitStructure.GPIO_OType = GPIO_OType_OD;
            GPIO_InitStructure.GPIO_PuPd = GPIO_PuPd_NOPULL;
            GPIO_Init(I2C1_SCL_GPIO_PORT, &GPIO_InitStructure);

            /*!< Configure sEE_I2C pins: SDA */
            GPIO_InitStructure.GPIO_Pin = I2C1_SDA_PIN;
            GPIO_Init(I2C1_SDA_GPIO_PORT, &GPIO_InitStructure);

tracernz, 2015-11-06:
It should be open-drain. If the intention was to use the 1 MHz current-source driver capability then that is configured by a call to SYSCFG_I2CFastModePlusConfig.

ledvinap, 2015-11-06:
This may be quite dangerous. Either STM32f303 hardware forces OD and ignores OType or devices on bus are strong enough to fight STM P-FET (possibly exceeding absolute maximum ratings)

tracernz, 2015-11-06:
It does say to configure open-drain in the driver docs so I assume it is does work: https://github.com/cleanflight/cleanflight/blob/master/lib/main/STM32F30x_StdPeriph_Driver/src/stm32f30x_i2c.c#L31
Haven't dug into the reference manual.

ledvinap, 2015-11-06:
I did not find anything in RM ...
I'll generate PR for this, hopefully it won't break anything ...

ledvinap, 2015-11-06:
Taulabs does use PP mode too (OD is used  on F4 targets): https://github.com/TauLabs/TauLabs/blob/12c5cec5bc6909d09947b0ffcaa51e62899a4066/flight/targets/sparky/board-info/board_hw_defs.c#L111

tracernz, 2015-11-06:
Good catch! I wonder if the config was copied from TL since I think the Sparky was the first F3 board for CF? The ST CPAL v2 I2C driver library configures the GPIO pins as open-drain as does any example I was able to find with a quick Google search. Of course open-drain is required by the nature of traditional I2C too so I'm confident that push-pull is wrong. Do you agree? It would be appreciated if you open an issue on the Tau repo as well, or I can do it, but it's nicer coming from the original discoverer. :-)

ledvinap, 2015-11-08:
@tracernz : It would be great if you can handle TauLabs.
for CF, #1472 is ready

hydra, 2015-11-28:
merged

closed 2015-11-28 by hydra