cleanflight/cleanflight #1473: Cleanflight needs better task scheduling

borisbstyle, 2015-11-06:
There is some serious jitter due to bad task scheduling. Some periodic parts of code can cause huge looptime variations. A good example of those is rx code and baro. Control loop should be as jitter free as possible.

@ledvinap 
I used time based scheduling in my branch with success to prevent jitter in control loop. Do you have any other suggestions to do this within the current code?

ledvinap, 2015-11-06:
Synchronizing loop to gyro reading (and using time of actual gyro sample) should help with one part of problem - input jitter effects should be minimal.

Second problem is jitter in command passed to ESC's. If jitter is problem, some form of queue may be implemented (it may be even shorter than one looptime), trading jitter for delay.

borisbstyle, 2015-11-07:
Yes i did it with using gyro sync as time reference. Reserving certain period of time before expected target time solves it.
Adding a queue upon that would be great to make sure all tasks get good attention.

ledvinap, 2015-11-07:
You don' need to synchronize gyro processing to gyro sampling. It is only necessary to use correct time in calculations (it is mostly delta time only, which is constant).
From outsize world only ESC output commands are visible. So if ESC command is sent (for example) 1ms after gyro sample time, you have this time to do the gyro reading and calculation. But it is irrelevant when you do it as long as the 1ms deadline is met.

This system can be implemented easily if all periodic tasks are short enough to fit in time left after per-loop calculations.

It would be great to time periodic tasks first and proceed based on results.

martinbudden, 2016-04-02:
@borisbstyle , do you think this issue can be closed now?

closed 2017-02-25 by hydra