cleanflight/cleanflight #1479: Blackbox Failsafe Event

borisbstyle, 2015-11-07:
@ProDrone 
It might be possible that after the new failsafe implementations the failsafe event is not being showed properly in blackbox?
Can you verify this? I have logs where I tested failsafe, but I dont see failsafe phase displayed properly?

ProDrone, 2015-11-15:
Strange.I just tried the same with latest CF at 80da7840064b331b5e6faf6466615762c486f035

Shown below:
Switched TX off (using X4RSB - CPPM mode)
`rxSignalReceived` goes down, followed by `rxFlightChannelsValid` shortly after.
Then after `failsafe_delay`, i see `failsafePhase` jump to 2 (LANDING). All as expected.
Can you please try again with latest?

[image]

borisbstyle, 2015-11-15:
stateflags is now small angle and I remember it going to failsafe before.

ProDrone, 2015-11-15:
These are the available state flags:
~~~c
typedef enum {
    GPS_FIX_HOME   = (1 << 0),
    GPS_FIX        = (1 << 1),
    CALIBRATE_MAG  = (1 << 2),
    SMALL_ANGLE    = (1 << 3),
    FIXED_WING     = (1 << 4),                   // set when in flying_wing or airplane mode. currently used by althold selection code
} stateFlags_t;
~~~

There is no failsafe stateflag.

Maybe you remember it wrong or mixed it up with `flightModeFlags`?
There i see: `ANGLE_MODE|undefined` while i expect something like `ANGLE_MODE|FAILSAFE_MODE`
This looks like a mismatch in typedef/defines between cleanflight firmware and blackbox viewer.

Anyway, i did not add these variables to the blackbox recorder, i think @thenickdude did, some time ago. I just recently have working blackbox hardware available.

ProDrone, 2015-11-15:
The RAW value of `flightModeFlags` is 1025 (1024 | 1) which is `FAILSAFE_MODE | ANGLE_MODE`, so that one is correct. The decoded value is wrong.

ProDrone, 2015-11-15:
@thenickdude
I modified the defines for Blackbox-log-viewer and made a pull request for that.
Works for me locally...

closed 2017-02-25 by hydra