cleanflight/cleanflight #1480: Feature Request: Custom Stick Map

OleEngr, 2015-11-08:
I've been using CF a few weeks with my first brushless quad. I came across a want/potential-need that might be addressed in CleanFlight to provide additional flexibility and versatility.  

I'm using an OrangeRX R720X receiver on DSMX.  This receiver will accommodate CPPM and PWM simultaneously, with the PWM channels being limited to the number of pins on the receiver (they are available with 6 to 12 channels of PWM).  All receivers will handle 12 channels of PPM regardless of the number of pins on the receiver.  For many users it is desirable to have the lightest receiver available that meets the needs of the model, hence the tendency to buy only as many PWM pins as you think necessary.

Since in my use the control sticks are PPM, I wondered if it was possible to move them to channels 8-12, or elsewhere, leaving the those 4 pins available for use as PWM channels. There isn't any problem for me to move the stick channels in my transmitter, a Devo 7e with Deviation FW. The only thing I have found in CF is the MAP command, and from what I can tell this just affects the assignment order of the first 4 channels to the sticks, not the channels they are attached to. 

 I have a quad with non-standard motor wiring, and a fellow came up with a custom motor mix command to set the motor position correctly for CF without having to rewire their location in a cable. I was hoping there might be a similar custom MAP function available for the control sticks. 

Sorry if I'm treading on ground that has already been turned over.  I barely know my way around and was lucky enough to find this forum.

ledvinap, 2015-11-08:
MAP works on first 8 channels. Move control channels to 5-8, use map 1234reta.

OleEngr, 2015-11-08:
I hadn't seen that command or realized that MAP used any but Channels 1-4.  Unfortunately, that is a specific solution, and not the general case solution I was looking for.  Also, I believe that solution conflicts with using channels 5 and 6 for in-flight tuning.  
By allowing the MAP to be positioned to channels 8-12, it places them firmly in PPM territory for receivers like the OrangeRX Rx20X, which have 12 channels PPM and up to 6-12 channels PWM depending upon the model ordered.  For an Orange R820X, with the MAP positioned at 9-12, the user has 8 channels of PWM available to them. and so on for the 6 and 7 channel versions.  The same receiver would have only 4 PWM channels available to the user for any of the MAP choices available today.

hydra, 2015-11-28:
@OleEngr Yes @ledvinap is correct.  Though perhaps allowing all channels to be remapped would be good, the problem here is the CLI gets in the way a little in the codebase due to the implementation of the map command.

owenduffy, 2016-07-15:
Without commenting on the proposed enhancement, my reading of the OrangeRX Rx20X manual is that the CPPM models all have 9 channels of PPM.

closed 2017-02-25 by hydra