cleanflight/cleanflight #1484: Constraint rcCommand in rate mode

gkvirkvia, 2015-11-11:
The problem I stumbled into is more of a mismatch of rate expo behavior in Configurator UI vs how things are actually implemented.

I'll start with a bit of backstory to my problem - I had to adjust my transmitters(Turnigy 9x - yeah I know :/) subtrim as midpoint for all of the controls seemed to be more around 1460 out of the box. This made my endpoints somewhat asymmetrical without further adjustments. So I tried to accommodate for that by ramping up my rate/expo so that the curve in Configurator UI hits ceiling before the end of X axis, believing that there's some kind of thresholding algorithm applied at the "ceiling" of that graph.
I was wrong, the quad ended up spinning way faster in one direction compared to other when I was hitting my sticks to the ends of their travel, both roll and pitch-wise. It took me some digging with UI and source code to confirm that there's no thresholding happening at this moment.

I think this behavior is somewhat confusing and needs to be fixed. I'm happy to provide patch as long as the maintainers are cool with what I'm proposing.
Let me know what you think guys, cheers!

SteveAmor, 2015-11-13:
If you use end point adjustments in your transmitter, that should fix your issue.

joshuabardwell, 2015-11-13:
I would suggest re-calibrating your 9x. It shouldn't be that far off when the sticks are centered.

gkvirkvia, 2015-11-13:
Thanks for replies guys, that's exactly what I ended up doing in the end and everything works fine. My point regarding the issue is more about how Cleanflights behavior is counter-intuitive, I think it still needs to be addressed.

joshuabardwell, 2015-11-14:
What's counter-intuitive, exactly?

gkvirkvia, 2015-11-15:
The fact that RC curves seem to be cut off at the top of the graph in Configurator, while in reality expo is applied even beyond this imaginary top boundary.

borisbstyle, 2015-11-27:
@gkvirkvia That's done on purpose and it's coming from multiwii. The expos on this way create more usefull curve. It's a bit different and steeper than TX expo.

Your main issue is your radio that it doesnt center perfectly

borisbstyle, 2015-11-27:
I believe you are refering to this where 2500 is used isntead of 2000 what you are expecting.
https://github.com/cleanflight/cleanflight/blob/d0b940bf756eef37a8fe3426d7ff1e5af6a705ed/src/main/io/rc_curves.c#L37

closed 2017-02-25 by hydra