cleanflight/cleanflight #1485: Aux configuration: "sum condition"

robustini, 2015-11-12:
It would be useful to have the ability to activate certain conditions only with the sum of Aux state, not like now that it only takes just one.
For example it might be useful to activate the "recovery beeper" using AUX only while disarmed.

joshuabardwell, 2015-11-12:
If you have a programmable transmitter, you can already do this, by programming all the switches on a single Aux channel, and adding their values together appropriately to cause the Mode to fall into certain ranges.

For example: Switch A causes channel to go from 0 to 100 us. Switch B adds to channel 1000 and 1500 us. The channel value will be either 0, 100, 1000, or 1600. This gives the full Boolean truth table for the two switches, and can be configured in the Modes tab accordingly.

For more on this topic:

http://forum.flitetest.com/showthread.php?18173-Taranis-advanced-programming-switch-multiplexing

robustini, 2015-11-12:
Hi Joshua, you're right, in fact you can do with the mixing.

closed 2015-11-12 by robustini