cleanflight/cleanflight #1489: Feature Request: Activate Buzzer after x seconds of inactivity

sticksen, 2015-11-13:
It would be nice to activate the buzzer after a crash, when all connection to the RC controller is lost. Maybe check the steering input and if nothing changes for x seconds, activate the buzzer?

joshuabardwell, 2015-11-14:
How is this different from the failsafe SOS beeping on RX loss?

Also, you can easily set a Mode to activate the buzzer, then use rxfail to configure an un-used aux channel to activate that mode when RX is lost.

sticksen, 2015-11-14:
Well, there are many levels of 'Failsafeness'.
1. The Rx quits its job completely. In this case, the SOS buzzing starts (which is good!). From what I assume, this is also the only case where rxfail starts to work, right?
2. The connection between Rx and Tx is lost. In this case, Cleanflight does nothing. 
But wait, I could set a failsafe on my Rx, which then sets an Aux channel to a certain value. :)

So yeah, it seems like I can fix this myself. Thanks!

closed 2015-11-14 by sticksen

joshuabardwell, 2015-11-14:
>2. The connection between Rx and Tx is lost. In this case, Cleanflight does nothing. 

This depends entirely on how the failsafe in your receiver is configured, for what it's worth. Failsafe in the receiver should be configured to either:

- No pulses
- Throttle channel goes below rx_min_usec
- Aux channel triggers mode "failsafe switch"

Any of these will invoke failsafe in CleanFlight when the RX loses link to the TX. 

If the RX is configured to hold last position, or to go to pre-sent conditions that do not match the description above, then CF will never go into failsafe, but this is a configuration error on the pilot's part, not a flaw in CF. CF cannot know that link is lost if the receiver is configured to obscure this fact.