cleanflight/cleanflight #1490: CJMCU support broken on betaflight 2.1.3

vasichkin, 2015-11-14:
I've compiled betaflight 2.1.3 and uploaded to my CJMCU.
After uploading i found:
It cannot connect to cleanflight configurator (timeout connecting).
Right after connecting to usb, motors start spinning.

I've tried betaflight 2.1.1 and it does not have thees issues.

SteveAmor, 2015-11-14:
I think you should report this issue in the betaflight repo https://github.com/borisbstyle/betaflight and not the cleanflight repo.  In which case, please close this issue.

SteveAmor, 2015-11-17:
@vasichkin please close this issue.  This is for reporting Cleanflight issues.

vasichkin, 2015-11-17:
@borisbstyle, i see no ability to report issue on your fork. Here i'm asking to close. Okay.

closed 2015-11-17 by vasichkin

borisbstyle, 2015-11-17:
CJMCU is not supported there :D. It's not on the list ;)