cleanflight/cleanflight #1499: flight stability

aerretechnology, 2015-11-17:
It's possible to auto calibrate the hovering middle position of the throttle stick when the battery discharge.
This features are good for example if you use 4s or 6s indifferent.
Sorry for my english.

digitalentity, 2015-11-17:
Do I understand correctly - you want to have a copter hovering at 50% throttle regardless of battery discharge?

aerretechnology, 2015-11-17:
Yes this is my idea.
For example the hovering value is 1400 ppm at 14v if the battery discharge
the ppm to mantein the hovering is to higher.
Then to replace 4s battery to an 6s the hovering value is to small and the
naze compensates the throttle value.

Inviato da AERRETECHNOLOGY mobile.

digitalentity, 2015-11-18:
It's not possible to do without altitude sensor (baro). And even with baro it will be tricky.

lostcontrol, 2015-11-18:
While not exactly what requested here, APM:Copter has something in this spirit to compensate for voltage drop: http://copter.ardupilot.com/wiki/current-limiting-and-voltage-scaling/#voltage_scaling

joshuabardwell, 2015-11-18:
Setting the ESCs to closed-loop mode may have a similar effect.

aerretechnology, 2015-11-18:
Ok testing the esc mode

jcowgar, 2015-11-18:
Maybe I am misunderstanding the original request, but isn't it a throttle curve? The way I understand the original request is not for voltage drop across a single battery, but is for when one uses a 3s for some flights and a 4s for other flights. For example, take an extreme case of flying with a 2s, The throttle may be 75% to hover then moving to a 6s may be at 5% to hover.

With PID banks, couldn't you configure the throttle curve and and throttle mid? Likely the PIDs would need adjusting for the various battery types anyway.

joshuabardwell, 2015-11-18:
I don't think Cleanflight currently has the ability to adjust the throttle curve like you are describing. You can adjust the amount of throttle expo and the mid-point of the expo curve. But you can't slide the entire throttle curve up and down like this.

bdoiron74, 2015-11-18:
My battery compensation code could be modified to do this. You'd probably
need hover voltage and hover throttle % parameters and then scale the
throttle input automatically based on battery voltage. Either by expo or
two-piece linear. I linearly scaled the whole thing, which just increased
the amount of dead space at the top as the battery depleted, but I don't
think that would be appropriate for cell count changes.

For the PIDs, you just need a base voltage and apply scaling accordingly. I
used the battery warning voltage and reduced the gain proportionally if the
battery was above that. You could run into resolution issues if you want to
handle huge voltage differences, but the floating point PIDs should be ok.
https://github.com/cleanflight/cleanflight/pull/1152

universam1, 2016-01-10:
+1
Eagerly waiting on PID scaled to voltage. Keep on good work!

bdoiron74, 2016-02-11:
@universam1  Keep an eye on this pull : https://github.com/cleanflight/cleanflight/pull/1805

It won't handle switching batteries, but I can see further development allowing it to. It would be pretty cool to not only adjust to battery drain, but also battery maxVoltage*cellcount. As-is, the top side of stick ends up being a dead zone as the battery drains, so it probably isn't the right way to handle different battery types. Remapping the stick throttle curve based on battery type might be a simple way around it though.

closed 2017-02-25 by hydra