cleanflight/cleanflight #1502: Couple of Issues after update

mattyra, 2015-11-17:
Naze Rev6 
200 Frame 
3 cell
BetaFlight/NAZE 2.1.3 Nov 13 2015 / 22:57:46 (1adcfaa)
LUXfloat Pid Controller 

Firstly Hello, and great work ! 

I have done the update and had the board flash with latest Betaflight, i have noticed a couple of issues,

Board does not save infomration correctly all the time, after setting horizon and level modes i could not remove them even after deleting via the configurator tab and saving with out they re apply the same setting have tried a couple of work throughs but same result, had to re flash to remove...

Next the Angle mode seem to always tilt to the left i have 3 naze Rev 6 boards and all do the same and yes i have calibrated them, i tuned the pids tight to see if this was a slap back on the I gain but not change...

Pretty much thats it i don't fly horizon or angle but i just noticed this.... as part of my testing...

Keep up the good work.... 

Austen

SteveAmor, 2015-11-18:
Can you please report betaflight issues on RC Groups and close this issue. This is for reporting Cleanflight issues.

borisbstyle, 2015-11-19:
You can create your issues here
https://github.com/borisbstyle/betaflight/issues

closed 2015-11-28 by hydra