cleanflight/cleanflight #1505: Cleanflight, plus-Mission-Control

sparrow082, 2015-11-20:
So instead of having cleanflight try to plan missions or deal with GPS and nav/cal, how about a stand alone board that dose just that. The purposes of the board would be to receive telemetry, determine it's position in 3D space, output standard control signals for yaw pitch roll altitude and acceleration. It would only have to know where it is and where it needs to go

The board shuld would have
Searil Tx, Rx,
Navigation calculator CPU
Locale RAM and/or SD card mimory
A battery 
GPS receiver module
Compass and/or barometer (if needed or to turn your acro F.C. into a deluxe F.C.)

I'd like to call this thing
plus-Mission-Control.

readerror67, 2015-11-20:
[image]

henn1001, 2015-11-20:
"Why though?" :D  

The main problem is that there is no code implemented yet that does gps nav with reasanable results. (that is being worked on though by DE). If space is your concern than a f4 fc is clearly the better and cheaper option. they are clearly capable of doing all that withouth relying on external boards.

If you want a extension board for scientific research where you can do a lot more than just nav stuff then there are probably better suited flight firmwares out there already. (apm & co)

tracernz, 2015-11-21:
Been done and abandoned: http://buildandcrash.blogspot.co.nz/2014/10/reflecting-on-four-years-of-development.html

ledvinap, 2015-11-21:
http://www.intel.com/content/www/us/en/do-it-yourself/edison.html ?

hydra, 2015-11-28:
closing, invalid.

closed 2015-11-28 by hydra