cleanflight/cleanflight #1508: Flip32+ magnetometer (HMC5883L) dissapears

ChrisStory, 2015-11-21:
I have a Flip32+ (http://www.amazon.com/gp/product/B0142Q57AA?psc=1&redirect=true&ref_=oh_aui_detailpage_o01_s00) that has a HMC5883L magnetometer. 

After I first flash the firmware to the board (v1.10, v1.08, and 11-18-2015 nightly) I am able to calibrate the magnetometer. When I unplug the board and then reconnect, the magnetometer 'disappears'. However, if I launch the CLI and then exit the CLI (causing the board to reboot), the magnetometer reappears.

hydra, 2015-11-28:
sounds like a hardware issue where the mag isn't getting enough power at startup.  try adding some delays in the mag initialisation code and report back once you've found something.

ashaposhnikov, 2015-12-16:
Hi guys,
I could confirm that after moving to the 1.11 magnetometer/compas 5883L is missing on my CJMCU board also. I'm using modified firmware though but I have the same behavior on original 1.11. Flashing to 1.10 immediately return magnetometer back.

Carl334, 2015-12-26:
I have almost exact same problem with remote monunted MAG;
My setup SPRacing F3 FC & UBLOX M8N GPS with compass.
I have the following configured and working. Configuration using CF 1.1.0 .
1. GPS
2. Bluetooth
3. Spektrum PPM with sat.
4. MWOSD
5. LED strip
6. Buzzer & voltage monitor
7. Other working sensors: Gyro, Accel & Baro
Removed HMC5883 MAG from FC board UBOX GPS module has HMC5883 MAG
Just prior to posting this I erased & re-flashed board.
CF Config. only recognizes MAG after I reboot; ie CLI 'status', 'exit'or changed a feature or almost anything that requires a reboot.
If unplug USB and reconnect no MAG if I reboot MAG! Only evidence of anything I can find is at bottom FC Config. screen and & CLI screen
is: I2C Error: or Errors: 2 or 4. After reboot I2C Errors = 0.
This is the best I describe problem this was the same condition before I reflashed with everything connected. This also happens with the battery connected.
I think that there needs to be some delays in the mag initialization code. That's way over my head.
I also uncovered the following: 5883 chip on SPR board 5983 chip on UBLOX board.

hankxx, 2016-01-02:
I have the Flip32 with external GPS magnetometer "Ublox 7 Series Mini GPS w/Compass  - the AP GPS" 
Testing DigitalEntity's iNav versions with CleanFilight 1.1.0.
Similar problem. When magnetometer correctly working it does not show up in Cleanflight. If magnetometer connected with SCL and SDA in i2c the other way around, MAG shows but quadcopter picture is not steady nor working regarding mag.

Xe54, 2016-09-27:
Have the same issue on SP Racing F3 Acro board with an external magnetometer (Ublox M8N GPS modul with compass). Two I2C errors are shown.

closed 2017-02-25 by hydra