cleanflight/cleanflight #1510: Invalid S.BUS decoding on PC

DreamNik, 2015-11-21:
Hi!
I've attached hardware S.BUS receiver to PC and tried to decode it.
(https://github.com/DreamNik/cleanflight/tree/x_plane)
(https://github.com/cleanflight/cleanflight/pull/1496)

The channel values were invalid. I've checked and looks like bit fields https://github.com/DreamNik/cleanflight/blob/master/src/main/rx/sbus.c#L108-L123 behave differently on different platforms.

I've created simple platform-independent decoder instead:
https://github.com/DreamNik/cleanflight/tree/fix_sbus_1

P.S. should I create new pull request for this?

ledvinap, 2015-11-22:
The struct bitfield  generates very fast code, notably on ARM with bitfield instructions. It will be probably better to fix structure declaration to make it work on PC.

Maybe just `unsigned syncByte:8`?

As for programatical bit extraction - on 32 bit machine it will be faster to read whole 32bit words and then handle bit extracting as mask / shift, refilling or joining or word boundaries.

DreamNik, 2015-11-22:
Well, the problem is that this bitfield IS platform dependent anyway.
C99 standard:
"If enough space remains, a bit-field that immediately follows another bit-field in a
structure shall be packed into adjacent bits of the same unit. If insufficient space remains,
whether a bit-field that does not fit is put into the next unit or overlaps adjacent units is
implementation-defined"

For example, on PC (MinGW,gcc) bits are packed this way:
11 bit field 1 ; 11 bit field 2 ; 10 bit padding ; 11 bit field 1 ; 11 bit field 2 ; 10 bit padding ; etc..

C/C++  tries to pack as many bitfields into "unsigned int" structure. It packs 2 of them, and adds 10 bit padding.

> Maybe just unsigned syncByte:8?
"unsigned" = "unsigned int" on 32 bit cpu, the same thing. Maybe you meant "unsigned short"?
I've tried that - packing becomes:
11 bit field 1 ; 5 padding ; 11 bit field 2 ; 5 bit padding ; etc..

All stuff like "__atribute((packed))__" , "#pagra pack..." doesn't affect this.

So I don't know current platform independent way to do this with bitfields.

DreamNik, 2015-11-22:
I'll setup toolchain for ARM target and check what code generated with bitfields and other methods. Then I'll choose fastest and/or smallest version.

ledvinap, 2015-11-23:
@DreamNik : The idea was to make syncByte bitfield too.

What compiler are  you using?

DreamNik, 2015-11-23:
> The idea was to make syncByte bitfield too.
Oh, I understood.
But this won't resolve issue.

> What compiler are you using?
gcc (i686-posix-dwarf-rev1, Built by MinGW-W64 project) 4.9.2

ledvinap, 2015-11-23:
maybe use attribute ((gcc_struct)) on windows?
https://gcc.gnu.org/onlinedocs/gcc/x86-Type-Attributes.html#x86-Type-Attributes

DreamNik, 2015-11-23:
This one did the trick!

DreamNik, 2015-11-23:
But I've already noticed that new version takes less size in code. Is it worth to analyze performance?

DreamNik, 2015-11-23:
Old one:
   text    data     bss     dec     hex filename
 118804     208   14144  133156   20824 ./obj/main/cleanflight_NAZE.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/cleanflight_NAZE.elf obj/cleanflight_NAZE.hex

./obj/main/NAZE/rx/sbus.o:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
 65 .text.sbusFrameStatus 00000138  00000000  00000000  00002434  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE

DreamNik, 2015-11-23:
New one:
   text    data     bss     dec     hex filename
 118668     204   14144  133016   20798 ./obj/main/cleanflight_NAZE.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/cleanflight_NAZE.elf obj/cleanflight_NAZE.hex

./obj/main/NAZE/rx/sbus.o:     file format elf32-littlearm

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
 65 .text.sbusFrameStatus 00000080  00000000  00000000  000025d8  2**2
                  CONTENTS, ALLOC, LOAD, RELOC, READONLY, CODE

DreamNik, 2015-11-23:
Well, maybe the speed is more important with new scheduler on the way.

ledvinap, 2015-11-23:
Personally I would prefer generic bitstream decoder (it may be used in other places eventually).
Using 32bit buffer and assuming that at most 32 bits may be extracted , it should be quite easy to implement..
That way frame decoding will be quite easy to read ..

DreamNik, 2015-11-23:
But at expense of cpu cycles...
Or it isn't as critical as I imagine?

ledvinap, 2015-11-23:
something like: 
```
typedef struct bitstream_s {
  unsigned *dptr;
  unsigned acc;
  unsigned acc_bits;
} bitstream_t;

uint32_t bitstream_extract(bitstream_t *bs, unsigned bits)
{
  uint32_t ret = 0;
  while(bits > 0) {
    if(!(bs->acc_bits)) {
      bs->acc = *bs->dptr++;
      bs->acc_bits = 8 * sizeof(*bs->dptr);
    }
    unsigned chunk = MIN(bits, bs->acc_bits);
    ret <<= chunk;
    ret |= bs->acc & ((1LL << chunk) - 1);
    bs->acc >>= chunk;
    bs->acc_bits -= chunk;
    bits -= chunk;
  }
  return ret;
}
```
It may optimize quite nicely ...

BTW: 32bit extraction should be special-cased on Intel, instead of 64bit shift trick; on ARM `(1 << 32) -1` will work correctly..

Little-endian is assumed.

DreamNik, 2015-11-23:
Won't this "caching" of 32-bit data cause any unwanted out-of-array reads?
It sounds unlikely, but just to be sure.

ledvinap, 2015-11-23:
Yes, it may read after array. Caller is expected to stop reading when actual buffer is exhausted, so stray data will be ignored. Padding bytes will be usually in this area ... 

Another problem is that aligned buffer start is assumed. It may be fixed with improved bitstream_t initialization (preload unaligned part into acc and set number of bits accordingly)

DreamNik, 2015-11-23:
WIP - Right now  I'm writing 4-byte aligned bitstream reader :)

DreamNik, 2015-11-23:
Should I create new branch and pull request for bitstream?

DreamNik, 2015-11-23:
Ok, introduced myself to unit testing.
Here's code:
https://github.com/DreamNik/cleanflight/tree/add_bitstream

closed 2017-02-25 by hydra