cleanflight/cleanflight #1512: S.BUS data race

DreamNik, 2015-11-22:
Hi!

I've noticed that if new S.BUS data arrives during "sbusFrameStatus" execution, then S.BUS channel data gets corrupted.

This is because new data is overwrites existing buffer on serial interrupt at any moment.
If that moment somehow take place, for example at https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/sbus.c#L194 , then half of data is overwritten.

This wouldn't a problem if reading 1 channel data is atomic. But it's not.

I think there is should be some kind of checks inside "sbusDataReceive" to skip data write if it's being processed right now.

tracernz, 2015-11-24:
This could well be related to all the problems with SBus dropouts. ;-)

borisbstyle, 2015-11-24:
@ProDrone 
This indeed explains a lot why we saw those dropouts after 1.9 upgrade.

This was implemented to prevent the glitching, but the real problem is in here.
https://github.com/cleanflight/cleanflight/commit/a64e2c4f1a6180a18cf3501bd980d8578c60a264

fnurgel, 2015-11-24:
> This wouldn't a problem if reading 1 channel data is atomic. But it's not.

Why isn't it atomic?

ledvinap, 2015-11-24:
There should be quite a lot of time (~4ms) to process received frame before next frame is received. 
What is causing such big delay (when armed)?
Are SBUS dropouts really caused by this?

borisbstyle, 2015-11-24:
@ledvinap 
It's a possible scenario and seen in the field if you read issues regarding that commit. It was specifically noticed between upgrade 1.9->1.10 specifically on slower F1 targets and faster looptimes.
Added timer might be hiding a real issue now?

Some reports:
https://github.com/cleanflight/cleanflight/issues/1395
https://github.com/cleanflight/cleanflight/pull/1340

But also reported on spektrum satellite here
https://github.com/cleanflight/cleanflight/issues/1394

borisbstyle, 2015-11-24:
Also based on some unexplained disasrms with SBUS reported and in the field reported by some this was what I want to add for some extra safety.
https://github.com/cleanflight/cleanflight/pull/1503

ledvinap, 2015-11-24:
@borisbstyle : Race described in this issue should be extremely rare. It can corrupt data only on channel that changes value (SBUS ARM channel value is fixed).
IMO this is inconsistent with reported problems you link ...

borisbstyle, 2015-11-24:
Maybe not the arm issue, but other channels can be corrupter and give bad values outside the range.

borisbstyle, 2015-11-24:
https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/rx.c#L491-L493

ProDrone, 2015-11-24:
> This was implemented to prevent the glitching, but the real problem is described in this issue.
https://github.com/cleanflight/cleanflight/commit/a64e2c4f1a6180a18cf3501bd980d8578c60a264

@borisbstyle 
The mechanism is still good to cover for real (radio-link) glitches. I think that any potential leak we discover should be addressed. I don't like the idea of making `sbusReadRawRC` atomic because there is floating point math in there, and it can be solved in a better way.

@ledvinap Extremely rare means it still happens, but just not as frequently, right?

ledvinap, 2015-11-24:
@ProDrone : My opinions is that there is another cause of glitches, not this bug (or at least not without another problem)

ledvinap, 2015-11-24:
The bug is much more serious ..

All bytes received after `SBUS_FRAME_SIZE` bytes up to `(SBUS_TIME_NEEDED_PER_FRAME + 500)` timeout are corrupting memory after `sbusFrame` ...

ProDrone, 2015-11-24:
@ledvinap
That is an ugly one! (i think we infringe Microsoft's buffer-overflow-patent here) :grimacing:
I'm on it right now...

borisbstyle, 2015-11-24:
The fix is easy.....just store it to another variable at the end

ProDrone, 2015-11-24:
@borisbstyle Hmm, don't get it. What code-lines and variables do you mean?

borisbstyle, 2015-11-24:
static sbusFrame_t sbusFrame, sbusFrameReceived;

// Receive ISR callback
static void sbusDataReceive(uint16_t c)
{
    static uint8_t sbusFramePosition = 0;
    static uint32_t sbusFrameStartAt = 0;
    uint32_t now = micros();

    int32_t sbusFrameTime = now - sbusFrameStartAt;

    if (sbusFrameTime > (long)(SBUS_TIME_NEEDED_PER_FRAME + 500)) {
        sbusFramePosition = 0;
    }

    sbusFrame.bytes[sbusFramePosition] = (uint8_t)c;

    if (sbusFramePosition == 0) {
        if (c != SBUS_FRAME_BEGIN_BYTE) {
            return;
        }
        sbusFrameStartAt = now;
    }

    sbusFramePosition++;

    if (sbusFramePosition == SBUS_FRAME_SIZE) {
        // endByte currently ignored
        sbusFrameDone = true;
        sbusFrameReceived = sbusFrame;
#ifdef DEBUG_SBUS_PACKETS
        debug[2] = sbusFrameTime;
#endif
    } else {
        sbusFrameDone = false;
    }
}

uint8_t sbusFrameStatus(void)
{
    if (!sbusFrameDone) {
        return SERIAL_RX_FRAME_PENDING;
    }
    sbusFrameDone = false;

#ifdef DEBUG_SBUS_PACKETS
    sbusStateFlags = 0;
    debug[1] = sbusFramereceived.frame.flags;
#endif

    sbusChannelData[0] = sbusFrameReceived.frame.chan0;
    sbusChannelData[1] = sbusFrameReceived.frame.chan1;
    sbusChannelData[2] = sbusFrameReceived.frame.chan2;
    sbusChannelData[3] = sbusFrameReceived.frame.chan3;
    sbusChannelData[4] = sbusFrameReceived.frame.chan4;
    sbusChannelData[5] = sbusFrameReceived.frame.chan5;
    sbusChannelData[6] = sbusFrameReceived.frame.chan6;
    sbusChannelData[7] = sbusFrameReceived.frame.chan7;
    sbusChannelData[8] = sbusFrameReceived.frame.chan8;
    sbusChannelData[9] = sbusFrameReceived.frame.chan9;
    sbusChannelData[10] = sbusFrameReceived.frame.chan10;
    sbusChannelData[11] = sbusFrameReceived.frame.chan11;
    sbusChannelData[12] = sbusFrameReceived.frame.chan12;
    sbusChannelData[13] = sbusFrameReceived.frame.chan13;
    sbusChannelData[14] = sbusFrameReceived.frame.chan14;
    sbusChannelData[15] = sbusFrameReceived.frame.chan15;

    if (sbusFrameReceived.frame.flags & SBUS_FLAG_CHANNEL_17) {
        sbusChannelData[16] = SBUS_DIGITAL_CHANNEL_MAX;
    } else {
        sbusChannelData[16] = SBUS_DIGITAL_CHANNEL_MIN;
    }

    if (sbusFrameReceived.frame.flags & SBUS_FLAG_CHANNEL_18) {
        sbusChannelData[17] = SBUS_DIGITAL_CHANNEL_MAX;
    } else {
        sbusChannelData[17] = SBUS_DIGITAL_CHANNEL_MIN;
    }

    if (sbusFrameReceived.frame.flags & SBUS_FLAG_SIGNAL_LOSS) {
#ifdef DEBUG_SBUS_PACKETS
        sbusStateFlags |= SBUS_STATE_SIGNALLOSS;
        debug[0] = sbusStateFlags;
#endif
    }
    if (sbusFrameReceived.frame.flags & SBUS_FLAG_FAILSAFE_ACTIVE) {
        // internal failsafe enabled and rx failsafe flag set
#ifdef DEBUG_SBUS_PACKETS
        sbusStateFlags |= SBUS_STATE_FAILSAFE;
        debug[0] = sbusStateFlags;
#endif
        // RX *should* still be sending valid channel data, so use it.
        return SERIAL_RX_FRAME_COMPLETE | SERIAL_RX_FRAME_FAILSAFE;
    }

#ifdef DEBUG_SBUS_PACKETS
    debug[0] = sbusStateFlags;
#endif
    return SERIAL_RX_FRAME_COMPLETE;
}

borisbstyle, 2015-11-24:
Didnt test it, but this is easier
[patch.txt](https://github.com/cleanflight/cleanflight/files/42853/patch.txt)

borisbstyle, 2015-11-24:
This is easier to read
[patch.txt](https://github.com/cleanflight/cleanflight/files/42854/patch.txt)

borisbstyle, 2015-11-24:
perhaps sbusFrameReceived should be made volatile?

borisbstyle, 2015-11-24:
Oh wait.. ledvinap mentioned this part

typedef union {
#    uint8_t bytes[SBUS_FRAME_SIZE];#
    struct sbusFrame_s frame;
} sbusFrame_t;

ProDrone, 2015-11-24:
Can you maybe make a PR to clearly show your proposed changes?

borisbstyle, 2015-11-24:
@ProDrone 
I was mixing the above issue and the one mentioned by ledvinap.

I am not fully familliar with SBUS protocol, but he assumes that there is a certain period SBUS_TIME_NEEDED_PER_FRAME + 500 where sbusFrame size is exceeded and all data will be corrupting something else what comes after sbusFrame[15]

borisbstyle, 2015-11-24:
Although there is some room reserved there
https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/sbus.c#L124-L132

ProDrone, 2015-11-24:
Yes there are several problems.
As first step (to prevent buffer overflow) i modified this one, please review/comment.

~~~C
// Receive ISR callback
static void sbusDataReceive(uint16_t c)
{
    static uint8_t sbusFramePosition = 0;
    static uint32_t sbusFrameStartAt = 0;
    uint32_t now = micros();

    int32_t sbusFrameTime = now - sbusFrameStartAt;

    if (sbusFrameTime > (long)(SBUS_TIME_NEEDED_PER_FRAME + 500)) {
        sbusFramePosition = 0;
    }

    if (sbusFramePosition == 0) {
        if (c != SBUS_FRAME_BEGIN_BYTE) {
            return;
        }
        sbusFrameStartAt = now;
    }

    sbusFrame.bytes[sbusFramePosition] = (uint8_t)c;

    if (sbusFramePosition < SBUS_FRAME_SIZE) {
        sbusFramePosition++;
        sbusFrameDone = false;
    } else {
        // endByte currently ignored
        sbusFrameDone = true;
#ifdef DEBUG_SBUS_PACKETS
        debug[2] = sbusFrameTime;
#endif
    }
}
~~~

borisbstyle, 2015-11-24:
shouldn't this be?
if (sbusFramePosition <= SBUS_FRAME_SIZE)

ProDrone, 2015-11-24:
> shouldn't this be?
if (sbusFramePosition <= SBUS_FRAME_SIZE)

No, that would put `sbusFramePosition` at `SBUS_FRAME_SIZE`
and allows access to `sbusFrame.bytes[SBUS_FRAME_SIZE]` on next iteration.

I made another mistake though, corrected below...

~~~C
// Receive ISR callback
static void sbusDataReceive(uint16_t c)
{
    static uint8_t sbusFramePosition = 0;
    static uint32_t sbusFrameStartAt = 0;
    uint32_t now = micros();

    int32_t sbusFrameTime = now - sbusFrameStartAt;

    if (sbusFrameTime > (long)(SBUS_TIME_NEEDED_PER_FRAME + 500)) {
        sbusFramePosition = 0;
    }

    if (sbusFramePosition == 0) {
        if (c != SBUS_FRAME_BEGIN_BYTE) {
            return;
        }
        sbusFrameStartAt = now;
    }

    if (sbusFramePosition < SBUS_FRAME_SIZE) {
        sbusFrame.bytes[sbusFramePosition] = (uint8_t)c;
        sbusFramePosition++;
        sbusFrameDone = false;
    } else {
        // endByte currently ignored
        sbusFrameDone = true;
#ifdef DEBUG_SBUS_PACKETS
        debug[2] = sbusFrameTime;
#endif
    }
}
~~~

ProDrone, 2015-11-24:
@borisbstyle I removed some of our comments above, to cleanup and make it easy to follow for other readers...

borisbstyle, 2015-11-24:
Maybe separate issue for this would be better as the original reported issue is something else actually?

I reviewed the code and it seems fine now. Going to do some testing in about an hour or 2 probably

ProDrone, 2015-11-24:
It does NOT work here, debugging.... but my time is up. I will continue later...

borisbstyle, 2015-11-24:
You did any debugging?

DreamNik, 2015-11-24:
Maybe just simple flag-based method will do. It prevents data overwrite while it's being processed at expense of dropping frame.
https://github.com/DreamNik/cleanflight/blob/fix_sbus_1/src/main/rx/sbus.c#L177

Sorry, I haven't made separate branch for this.That commit related to other issues:
https://github.com/cleanflight/cleanflight/issues/1511
https://github.com/cleanflight/cleanflight/issues/1510

borisbstyle, 2015-11-24:
@ProDrone 
I think this might be your issue. (This wrights to 0 and increments)
'''
        sbusFrame.bytes[sbusFramePosition++] = (uint8_t)c;
'''

Should be like below. This increments and wrights
'''
        sbusFramePosition++;
        sbusFrame.bytes[sbusFramePosition] = (uint8_t)c;
'''

borisbstyle, 2015-11-24:
Also this means that it will mark sbusFrameDone during next receive. It should be == SBUS_FRAME_SIZE instead of <

    if (sbusFramePosition < SBUS_FRAME_SIZE) {
        ....
        sbusFrameDone = false;
    } else {

ledvinap, 2015-11-24:
@prodrone: your code writes last byte, but tries to set done on next received byte (which does not arrive)

```
    if (sbusFramePosition < SBUS_FRAME_SIZE)
        sbusFrame.bytes[sbusFramePosition++] = (uint8_t)c;
    if (sbusFramePosition < SBUS_FRAME_SIZE) {
        sbusFrameDone = false;
    } else {
        sbusFrameDone = true;
    }
```

borisbstyle, 2015-11-24:
@ledvinap 
Yes exactly what I found. It's so easy to overlook stuff :D

borisbstyle, 2015-11-24:
And increment should happen prior to write

ledvinap, 2015-11-24:
Why? First stored byte is `SBUS_FRAME_BEGIN_BYTE` at position 0.

borisbstyle, 2015-11-24:
O right....yeah i was comparing to the old code where the right was in the beginning of the function indeed.

borisbstyle, 2015-11-24:
But maybe we still want to write stuff after the last channel? Now being unused, but there might be additional data sent?

ledvinap, 2015-11-24:
@borisbstyle : Done flag is set when 25bytes  are received. Processing may start immediately, before next byte is received.
It is possible to handle next byte as error and discard frame, but that will be non-deterministic (identical frames will be handled differently based on sbusFrameStatus call timing)

borisbstyle, 2015-11-24:
So to address both issues. Also one mentioned above.

    static sbusFrame_t sbusFrame;
    static volatile bool sbusFrameProccessing;
    
    // Receive ISR callback
    static void sbusDataReceive(uint16_t c)
    {
        static uint8_t sbusFramePosition = 0;
        static uint32_t sbusFrameStartAt = 0;
        uint32_t now = micros();
    
        int32_t sbusFrameTime = now - sbusFrameStartAt;
    
        if (sbusFrameTime > (long)(SBUS_TIME_NEEDED_PER_FRAME + 500)) {
            sbusFramePosition = 0;
        }
    
        if (sbusFramePosition == 0) {
            if (c != SBUS_FRAME_BEGIN_BYTE) {
                return;
            }
            sbusFrameStartAt = now;
        }
    
        if (sbusFramePosition < SBUS_FRAME_SIZE) {
            if (!sbusFrameProccessing) {
                sbusFrame.bytes[sbusFramePosition] = (uint8_t)c;
            }
            sbusFramePosition++;
        }
        if (sbusFramePosition < SBUS_FRAME_SIZE) {
            sbusFrameDone = false;
        } else {
            sbusFrameDone = true;
    #ifdef DEBUG_SBUS_PACKETS
            debug[2] = sbusFrameTime;
    #endif
        }
    }

borisbstyle, 2015-11-24:
uint8_t sbusFrameStatus(void)
    {
        if (!sbusFrameDone) {
            return SERIAL_RX_FRAME_PENDING;
        }
        sbusFrameDone = false;
        sbusFrameProccessing = true;
    
    #ifdef DEBUG_SBUS_PACKETS
        sbusStateFlags = 0;
        debug[1] = sbusFrame.frame.flags;
    #endif
    
        sbusChannelData[0] = sbusFrame.frame.chan0;
        sbusChannelData[1] = sbusFrame.frame.chan1;
        sbusChannelData[2] = sbusFrame.frame.chan2;
        sbusChannelData[3] = sbusFrame.frame.chan3;
        sbusChannelData[4] = sbusFrame.frame.chan4;
        sbusChannelData[5] = sbusFrame.frame.chan5;
        sbusChannelData[6] = sbusFrame.frame.chan6;
        sbusChannelData[7] = sbusFrame.frame.chan7;
        sbusChannelData[8] = sbusFrame.frame.chan8;
        sbusChannelData[9] = sbusFrame.frame.chan9;
        sbusChannelData[10] = sbusFrame.frame.chan10;
        sbusChannelData[11] = sbusFrame.frame.chan11;
        sbusChannelData[12] = sbusFrame.frame.chan12;
        sbusChannelData[13] = sbusFrame.frame.chan13;
        sbusChannelData[14] = sbusFrame.frame.chan14;
        sbusChannelData[15] = sbusFrame.frame.chan15;
    
        if (sbusFrame.frame.flags & SBUS_FLAG_CHANNEL_17) {
            sbusChannelData[16] = SBUS_DIGITAL_CHANNEL_MAX;
        } else {
            sbusChannelData[16] = SBUS_DIGITAL_CHANNEL_MIN;
        }
    
        if (sbusFrame.frame.flags & SBUS_FLAG_CHANNEL_18) {
            sbusChannelData[17] = SBUS_DIGITAL_CHANNEL_MAX;
        } else {
            sbusChannelData[17] = SBUS_DIGITAL_CHANNEL_MIN;
        }
    
        if (sbusFrame.frame.flags & SBUS_FLAG_SIGNAL_LOSS) {
    #ifdef DEBUG_SBUS_PACKETS
            sbusStateFlags |= SBUS_STATE_SIGNALLOSS;
            debug[0] = sbusStateFlags;
    #endif
        }
        if (sbusFrame.frame.flags & SBUS_FLAG_FAILSAFE_ACTIVE) {
            // internal failsafe enabled and rx failsafe flag set
    #ifdef DEBUG_SBUS_PACKETS
            sbusStateFlags |= SBUS_STATE_FAILSAFE;
            debug[0] = sbusStateFlags;
    #endif
            // RX *should* still be sending valid channel data, so use it.
            return SERIAL_RX_FRAME_COMPLETE | SERIAL_RX_FRAME_FAILSAFE;
        }
    
        sbusFrameProccessing = false;
    
    #ifdef DEBUG_SBUS_PACKETS
        debug[0] = sbusStateFlags;
    #endif
        return SERIAL_RX_FRAME_COMPLETE;
    }

digitalentity, 2015-11-24:
Why not get rid of the callback from the ISR and process incoming data
asynchronously? Let serial code buffer the data.

borisbstyle, 2015-11-24:
@digitalentity 
I would prefer as many as possible via ISR, especially these serial protocols. It's there why not use it :D

digitalentity, 2015-11-24:
Think cpu time scheduling. Isr can't be delayed, conventional code can be
scheduled to run when cpu is free. We want our isrs as fast as possible to
get minimal jitter.

ledvinap, 2015-11-24:
As for the race ..  IMO it is almost impossible to trigger on real hardware.

sbusFrameDone is set when all 25 bytes are received and cleared when SBUS_FRAME_BEGIN_BYTE is received after timeout.

Multiple receive interrupts needs to be handled between checking sbusFrameDone in main loop code and finishing frame processing to trigger this bug.

ledvinap, 2015-11-24:
@digitalentity : Serial receive buffer would be great, but the capability  to resync after interrupt is important and is timing critical. It may be a bit hard to handle the timing correctly from main loop. (It would be trivial with RTOS ;-) )

ledvinap, 2015-11-24:
@DreamNik : I am afraid you will need thread synchronization primitives to handle all race conditions on multitasking OS ...

DreamNik, 2015-11-24:
> I am afraid you will need thread synchronization primitives to handle all race conditions on multitasking OS ...
In general - yes. But in this case we have 1 ISR writer and 1 main loop reader.
So, I think it's sufficient to use simple flag for this.

P.S. I'm testing on PC this and my method doesn't work on PC, like you said, because I run multiple threads simultaneously. But on MCU we have only one cpu thread and thus there are no simultaneous processes - only ISR can interrupt at any moment.

DreamNik, 2015-11-24:
> As for the race .. IMO it is almost impossible to trigger on real hardware.

But thankfully (or unfortunately) like piece of cake on PC :)

ledvinap, 2015-11-24:
@borisbstyle : your code is still theoretically incorrect:
- `if (!sbusFrameDone) {` - sbusFrameDone is set
- few bytes are received, partially overwriting buffer
- `sbusFrameProccessing = true;`

ledvinap, 2015-11-24:
IMO it is reasonable to fix writing after buffer and assume that main loop code will always execute at least for few us between serial receive interrupts (sbus characters are 100us apart)

borisbstyle, 2015-11-24:
@ledvinap Yes you are right. Well anyway I realize that @DreamNik is doing something completaly else on a completaly different platform.

@DreamNik May I ask what exactly you are working on? I see many issues from your side related with PC platform. Are you building a PC flight controller or something?

DreamNik, 2015-11-24:
In short - I'm building Cleanflight to run on PC with all the hardware being emulated by flight simulation software.

DreamNik, 2015-11-24:
I've put detailed description at https://github.com/DreamNik/cleanflight/blob/x_plane/sil/docs/description.md
It's working but still WIP.

borisbstyle, 2015-11-24:
@DreamNik 
Very interresting project. What is your main motivation behind it?

borisbstyle, 2015-11-24:
@ledvinap 
Can you have a quick scan through spectrum rx code as it looks very simillar. I know we had even more reports from spektrum glitches

https://github.com/cleanflight/cleanflight/blob/master/src/main/rx/spektrum.c#L95-L115

DreamNik, 2015-11-24:
https://github.com/borisbstyle
I've updated firmware on my quadcopter and found no old crazy autotune mode.
But there is new G-Tune mode.
And since my quad not ready for flight right now, I've decided that I can actually see how it works by attaching Cleanflight to simulator :)

ledvinap, 2015-11-24:
@DreamNik : Your best chance it make sure that interrupt emulating threads are not interrupted by main code. Scheduler policy may help here. It won't fix this, but at least some CF assumptions will be true (interrupt code is assumed to run atomically from main loop perspective)

Using atomic block will be probably easiest .. maybe some trick similar to ATOMIC_BLOCK may be used. Single macro may be used to make current scope atomic (enter atomic state and declare variable with cleanup that will leave it). And that just hunt places that have timing issue ...

DreamNik, 2015-11-24:
@ledvinap 
I know, but I don't want to mess current code with simulation code.
Mutexes and other stuff will be confusing (right now) for main code.

> Your best chance it make sure that interrupt emulating threads are not interrupted by main code.

I thought of 2 options:
1 - create in main code single annex function that'll handle all ISR emulation
2 - pause main thread, execute ISR emulation, resume thread - but it's Windows-only method

P.S. maybe we should move discussion about simulation into different comment thread? It's getting messy here.

[4 more replies]

closed 2015-11-28 by hydra