cleanflight/cleanflight #1516: Led strip issue

joeynuggetz, 2015-11-24:
I have been experiencing some issues with mapping the indicator function on certain leds in a strip. Using a 6 led strip I have been experiencing intermittent problems where the leds on the ends of the strip fail to light up. Not sure if this is related to my strip or not. I ordered a strip of assignable leds and cut off a section of 5 or 6 and once I added them to the chain the assignments were more reliable and was able to get the intended LEDs to function as indicators. Here is the original strip and the set I got to test adding more to the chain. 
http://amzn.com/B0140W5AEE    Im not using the buzzer
http://amzn.com/B00JR0MJIU     These suck by the way.

add1ct3dd, 2015-11-24:
Sounds like a hardware issue?

joeynuggetz, 2015-11-24:
Not sure. I don't recall having any issue setting led color. Just the indicator function at times. Will do more testing tonight.

hydra, 2015-11-28:
hardware issue, please post on forums first for support.

closed 2015-11-28 by hydra