cleanflight/cleanflight #1520: Brushed motor support does not do 100% duty cycle

geebles, 2015-11-25:
With the current brushed motor support in cleanflight, it appears to just reduce the PWM minimum value only (so that when "Minimum Throttle" is set to 1000, PWM is set to 0%) and the top end is unaffected, meaning that when you set "Maximum Throttle" to 2000, what is actually happening is the PWM is 'on' for '2000 time' and then 'off' for about '250-500' time (this applied to framing the PWM for ESCs, but for brushed it means that full throttle isn't actually 100% PWM duty.

As a quick hack I set "Maximum Throttle" to 2500, which according to the scope shows 100% duty cycle (hooray) however there is some weirdness in the PID controllers because when at full or very high throttle the copter will lean to one side (the acc/gyro is reading this but is ok with it, so the PID controller is making it lean).. reducing back to 2000 and full throttle is vertical as expected.

With more brushed FC's coming out, this would be really good to fix! :) PS I haven't actually looked at the motor/PWM code so my observations are purely from playing with config values and measuring duty cycle with a scope.

SteveAmor, 2015-11-25:
Looks like this was also confirmed in PR #1005

hydra, 2015-11-28:
yes, it's the same. there was never a proper fix for it.  thanks for reminding us all that this still needs to be done.

alcurtis93, 2015-11-30:
Please fix this. There's a lot of us brushed guys now running 32bit boards and they're already underpowered enough without issues like this ;)

ghi001, 2015-11-30:
Same here ! Waiting for it !!!!

LukeRitchie, 2015-12-07:
Yes agree, this needs fixing. I wait patiently. At the moment its forcing people to not buy 32 bit flight controllers and use multiwii, so this is a downside for the cleanflight community. Is there an easy fix, could we talk to borisb about putting it into betaflight?

JohnOCFII, 2015-12-07:
Just started a brushed Quad build for some indoor winter flying.  Another vote for this fix.

goebish, 2015-12-07:
Now I understand why the cjmcu is underpowered.
Another vote :)

joeynuggetz, 2015-12-09:
I'm a brushed guy too. Would love to see this fix.

matrixFLYER, 2015-12-10:
pls a fix thank you

Pedro147, 2015-12-10:
Having just bought a Scisky myself I would love to see this issue resolved sooner rather than later. Like others have said here, there are more and more 32 bit boards coming out regularly and this will slow down improvements in the sector

SteveAmor, 2015-12-10:
I hope to have a look at this :)

SteveAmor, 2015-12-11:
No pressure then ;-)

Pedro147, 2015-12-12:
No pressure at all SteveAmor. You have 48 hrs :)  Got Ya

SteveAmor, 2015-12-13:
I've had a really good look at this and can't find the issue you guys are having.  
Note that I've had to look at the output of a Naze32 on an analogue oscilloscope (not flown as I don't have a brushed craft).
I am using the latest release in Cleanflight Configurator (1.11.0 NAZE 2015-11-28 17:37 (stable)).
Tried with both motor_pwm-rate = 8000 and = 16000.
When the FC is not armed, there is no PWM signal.  If I ```set max_throttle = 2000``` it will result in 100% duty cycle at full throttle.  I also needed to increase my throttle end point so that Configurator shows 1942 usecs to get the PWM output to 100%.
Note, during my testing, I had P=4, I=0, D=0.
Did I miss the issue here?

alcurtis93, 2015-12-13:
I really think this needs looking at by someone with a brushed board to hand as I personally own a scisky 32 bit board from banggood and it's definitely down on power compared to something like a hubsan fc. Really appreciate you looking though Steve. The original OP has tested it with his own board with an oscilloscope and the issue is definitely there

geebles, 2015-12-13:
Let me double check this evening using latest version!

joshuabardwell, 2015-12-13:
If you are using the Motors tab to generate the outputs, then PID gains don't matter.

geebles, 2015-12-13:
I had spotted this issue in very early development of my brushed board, and it appears that since early development this was fixed (somehow) as I am unable to reproduce and what SteveAmor said is 100% true and I am able to reproduce that on my brushed board! 2000 motor output is 100% duty cycle, upping to 2500 actually does nothing (as soon as it hits 2000 it is 100%, so actually causing more issues).. it is odd as I double checked and did the 2500 fix at the time and could prove it working! Oh well..

Closing this issue as it isn't actually an issue, I should have re-checked before creating it in the first place! Thanks everyone for helping and looking into it!

closed 2015-12-13 by geebles