cleanflight/cleanflight #1522: Crash Sense Feature

Ozzy33, 2015-11-26:
What about a feature that could detect a crash and immediately stop motors and or disarm to prevent motor/ESC damage? Blackbox data should give good feedback on crashes data. Of course it should be an option to enable it and set motor stop or disarm. Priority = ultra low.

digitalentity, 2015-11-26:
That would be a super cool thing to have, however this should have zero false positives if you don't want your quad to disarm inflight on its own.

borisbstyle, 2015-11-27:
Good configured failsafe or manual disarming does the same thing

ProDrone, 2015-12-03:
I had the same idea in my mind for about a year, then i saw the discussion about it here: http://www.rcgroups.com/forums/showthread.php?t=2464844&page=310#post32853731

When it is implemented it should be as reliable as our airbag systems, maybe requiring dedicated sensors. I think it will turn out to be the one feature that is always turned off.. Or it will be used as a scapegoat for unexplained crashes (well, that may come in handy ;-)

Yamaford, 2015-12-19:
Looking at my blackbox logs its pretty easy to see a crash. All 3 Gyro's go completely nuts. That would be pretty hard to duplicate on purpose while doing acro maneuvers.
[image]

Attached you can see a lot of gyro action, throttle is below min check and there are no RC commands on any axis. That would be a good time to stop the motors.

closed 2017-02-25 by hydra