cleanflight/cleanflight #1523: How can this part of the code be accepted by GNU compiler?

wz233, 2015-11-27:
In the file "timer.c" (src/main/drivers/timer.c), I found following code:

```C
        switch(bit) {
            case __builtin_clz(TIM_IT_Update): {

                if(timerConfig->forcedOverflowTimerValue != 0){
                    capture = timerConfig->forcedOverflowTimerValue - 1;
                    timerConfig->forcedOverflowTimerValue = 0;
                } else {
                    capture = tim->ARR;
                }

                timerOvrHandlerRec_t *cb = timerConfig->overflowCallbackActive;
                while(cb) {
                    cb->fn(cb, capture);
                    cb = cb->next;
                }
                break;
            }
            case __builtin_clz(TIM_IT_CC1):
                timerConfig->edgeCallback[0]->fn(timerConfig->edgeCallback[0], tim->CCR1);
                break;
            case __builtin_clz(TIM_IT_CC2):
                timerConfig->edgeCallback[1]->fn(timerConfig->edgeCallback[1], tim->CCR2);
                break;
            case __builtin_clz(TIM_IT_CC3):
                timerConfig->edgeCallback[2]->fn(timerConfig->edgeCallback[2], tim->CCR3);
                break;
            case __builtin_clz(TIM_IT_CC4):
                timerConfig->edgeCallback[3]->fn(timerConfig->edgeCallback[3], tim->CCR4);
                break;
        }

```

"__builtin_clz" is used to count leading zeros of a variable (or constant), but how can this function be implemented in a constant expression? Because inside "switch case", each case should be constant, if you use a function inside "swtich - case", the compiler should give you errors saying:
 "**expression must be an integral constant expression**" and 
"**function call is not allowed in a constant expression**".

Anyone can help me understand this?

Thanks!

ledvinap, 2015-11-27:
gcc does accept expression if it can evaluate it at compile time. 
The code can be replaced with version that uses arithmetic operators (https://github.com/ledvinap/cleanflight/blob/local/src/main/common/utils.h#L55), but this seemed cleaner.

The `__builtin_` version may  be hidden in function and implemented for non-gcc/clang compilers.
This particular code is performance critical, each instruction has noticeable effect on performance ..

wz233, 2015-11-27:
Thanks for your reply @ledvinap . KEIL cannot evaluate the value at compile time. ARMCC has a function "__CLZ", but it cannot be used in "switch case" even if the number is constant. Sad~~:(

closed 2015-11-27 by wz233

ledvinap, 2015-11-27:
@wz233 : Just use the macro I linked for labels and `__CLZ` to get value of `bit`.