cleanflight/cleanflight #1524: Quadcopter drifting to the side no matter which setting is used in stable/horizon mode!

mskrzetuski, 2015-11-28:
Dear cleanflight creators,

I recently started with the quadcopter as a hobby and purchased my first FPV quadcopter to try things at home. I wanted to calibrate everything in cleanflight and use it in stable/horizon mode because this is the only mode I am currently able to fly in. With my Blade Nano QX I never experiences problems like this.

Directly after start up the quadcopter starts turning to the right very strongly (the more throttle is applied the worse it gets) and trimming on the transmitter is not helping at all even after about 50 pushes in the opposite direction. It seems that the accelerometer 'thinks' that the copter is tiled 45 degree to the left.

The quad copter I have is DYS X160 which can be purchased at http://www.rctimer.com/product-1464.html. I tried many settings including the ones from http://www.rctimer.com/product-1464.html and of course I calibrated the accelerometer and set up modes inside cleanflight. Resetting everything to default values does not work out either and causes the quadcopter to oscillate like crazy.

I am totally lost and am afraid that I wasted money on that carbon dragon.

Could you please help me debug this?

Cheers
Maciej

tigert, 2015-11-28:
Are you trimming on the transmitter (bad idea) or with cleanflight? (unarm, move throttle to max, yaw center and poke the stick a bunch of pokes to the opposite direction to what the quad is drifting to, the flight controller probably blinks the led to acknowledge the settings) - every now and then arm and see if it is leveling better.

I had to do this after a crash, and I do it every time it drifts off noticeably from level hover. Wind will push it around and no trim can fix that, but that is a different thing.

hydra, 2015-11-28:
for support please use forums not issue tracker.  check all your props and motors are aligned so the motor shafts point straight up, especially when using tubular arms.

closed 2015-11-28 by hydra

hankJesus, 2020-02-27:
Sorry to comment on such a dead thread.
Following tigert's comment, what does this change on the flight controller? I've tried everything, and this fixed my meteor65's drift. 
Thanks!