cleanflight/cleanflight #1528: Why was Harakiri support removed?

seattleite09, 2015-11-30:
Harakiri was one of the controllers I was using. It feels distinctively different from MW23/Rewrite/Luxfloat, and used to work very well in Horizon as well. I don't fully understand why it was removed???

seattleite09, 2015-11-30:
"Removed less popular PID controllers - experimentation time is over."

"Experimentation" time is over? Hmmm, sounds like a sweeping assumption + single-handed decision; but I guess you are the owner, and this is not a democracy.

A bit disappointing.

digitalentity, 2015-11-30:
There was a discussion on this topic https://github.com/cleanflight/cleanflight/pull/1012
Also @borisbstyle tested effect of removal of some PID controllers in his BetaFlight on a very broad group of pilots, myself included (I used to fly Harakiri). To me it's not a single-handed decision

seattleite09, 2015-11-30:
Yes, this topic was discussed in #1012 and there was evidence to suggest that Harakiri was still being used, that it was stable, and that it flew very well on a multitude of platforms -- I have been using it on my TBS Discovery, where it performs beautifully and flies better than Luxfloat and PIDC1 with a more natural "feel". In spite of that evidence though, it seems to me that a single-handed decision was still made to remove PIDC5, as a result of Hydra's preference to only "keep PIDC 1,2,3 and remove the rest". It's fine, I am just pointing out the facts. The difference between Harakiri and PIDC3 is not that significant, I guess I can downgrade back to PIDC3... but frankly I always considered PIDC5 superior to PIDC3. Not sure why the latter was kept over the former.

seattleite09, 2015-11-30:
...and by the way, you mentioned Boris' Betaflight and the discussion there. Boris' work is great, but several BetaFlight releases failed several times on my 250 / F1 setup -- possibly due to overclocking and other unsafe optimizations; so even though some have had amazing success with it, others -- myself included -- have found it impossible to set up. So I hardly consider it a benchmark here. I haven't tried the latest release, but that's beside the point. The fact that PIDC5 is missing from BF is not a sufficient reason to remove it from CF.

digitalentity, 2015-11-30:
I believe a sufficient reason for removing **some** PID controllers is memory requirements to move on with the development of other features. Harakiri was unlucky to be one of two least used controllers. Although actual usage statistics were never calculated, you can pick any video from Youtube featuring Cleanflight and if setup is specified it would almost always read PID controller 0,1 or 2.

joshuabardwell, 2015-11-30:
>Not sure why the latter was kept over the former.

I don't have any special insight into Hydra's thoughts, but my guess would be far greater adoption.

seattleite09, 2015-11-30:
Yeah. Making room for other features would be a good reason, I guess. I will just switch to Lux on all my quads, I guess.

closed 2015-11-30 by seattleite09

borisbstyle, 2015-11-30:
@seattleite09 
It has nothing to do with betaflight. It is the maintainability. Even 3 pid controllers is still much. 
I just basically did a "popularity research". PID0, 4 and 5 were by far least popular.