cleanflight/cleanflight #1530: Flash - Add ability to store configuration or other data in onboard flash chip

digitalentity, 2015-12-01:
Currently onboard m25p16 flash chip is used only for blackbox storage. Why not reserve first blocks of flash for configuration storage?
Benefits:
1) Free up 2kb of MCU's flash memory on F1 targets for something useful
2) Be able to increase configuration storage to remember something useful - i.e. waypoints for autonomous mission.

digitalentity, 2015-12-01:
@thenickdude, you are the author of flashfs, do you think this is doable without much effort?

thenickdude, 2015-12-01:
@digitalentity it wouldn't be worth moving system configuration there because we'd need an alternate mechanism for the majority of targets which don't have a dataflash chip onboard.

It's doable from a technical perspective.

digitalentity, 2015-12-01:
From config.c perspective it would still be _mostly_ the same - calls to writeEEPROM(), readEEPROM(), isEEPROMContentValid() and calculateChecksum() would be replaced on a per-target basis.

I'm asking if this is doable mostly with waypoint navigation in mind. It would be great to setup a mission at home, go to the field and execute, but for that I'd need a non-volatile configuration storage. Onboard flash chip looks the best option so far.

lostcontrol, 2015-12-01:
@digitalentity Sorry, this is off-topic but is it really worth adding features like waypoint navigation? I guess this would make the code more complex and yet you wouldn't have 1/10 of the nice features available in e.g. Ardupilot. I'm using Cleanflight since ~1 month so I'm new here but I switched to it because I wanted something simple for a 250. Please, keep Cleanflight simple!

digitalentity, 2015-12-01:
@lostcontrol There is some demand for better navigation - #197, #270, #476, #530, #843. Waypoints go along with better navigation and RTH capabilities, why not have them if we would have better GPS/RTH?

lostcontrol, 2015-12-01:
@digitalentity Well, to keep it short, I would personally go for Ardupilot for anything GPS-based. Waypoint navigation supports many features (splines, circle, change speed, altitude, trigger camera, ROI, etc) and Tower on your Android smartphone makes it really handy to use. Just thought autonomous flighting was not a focus for Cleanflight (well, except RTH).

thenickdude, 2015-12-01:
The thing about waypoint navigation is that we have to support almost all
of it in order to support simple return-to-home GPS, which is very much a
goal for Cleanflight. Adding waypoint navigation on top of position hold
and RTH is not a big task.

digitalentity, 2015-12-01:
Exactly my point. Thanks for clarifying this @thenickdude!

skaman82, 2015-12-01:
@lostcontrol I think adding waypoint functionality doesn't necessary makes Cleanflight more complicated if its done right. I also think it would be not the right way to just copy Ardupilot functionality with all its futures and complexity. A good example are the new advanced consumer drones. Here you can setup you Waypoint settings witch a few klicks or tips on your mobile device. This shows that setup for  autonomous flight functionality dosen't have to be like rocket science. The less you let the user setup the easier and more fun to use it will be.

lostcontrol, 2015-12-01:
Yes, actually RTH is waypoint navigation, agreed. I just think that autonomous flight is useful for video, picture, mapping and such things that quickly need more advanced features. As long as you don't make Cleanflight another Ardupilot, it is fine :wink: KISS

digitalentity, 2015-12-01:
@thenickdude can you give some insite on how to reserve i.e. 1K of spi flash for arbitrary data storage?

thenickdude, 2015-12-01:
@digitalentity You just want `tailAddress` to start at at least 1kB at FC init, rather than the 0 it currently starts at. That way, flashfs file allocations will always ignore the first 1kB. This could be accomplished by setting the initial value of the `left` variable to `1` in `flashfsIdentifyStartOfFreeSpace()`, which would exclude the first 2kB of flash space from normal FlashFS allocations. Then bypass FlashFS and call the m25p16 routines directly to [re]write the first 2kB of flash for configuration storage. 

Though I would really suggest that you never move configuration out of built-in flash. Every target needs configuration storage, but only the fancy ones with outboard SPI flash can support waypoint navigation. So store the waypoint data on the SPI flash. Now you only need one persistence mechanism for configuration data (inbuilt flash) which is common to all targets, and only one persistence mechanism for waypoints (SPI flash).

digitalentity, 2015-12-01:
@thenickdude Thanks. I'm not going to move config to SPI flash at the moment. I'll use SPI flash (if available) for persistence waypoint data storage.
Though I don't want to create another abstraction layer besides flashfs. How about I create an extension to flashfs in form of few functions:
readPersistenceStorageBlock(int blockNum, void * buf)
writePersistenceStorageBlock(int blockNum, void * buf)
to access blocks of predefined size (i.e. 256 bytes) in persistence storage?

closed 2017-02-25 by hydra