cleanflight/cleanflight #1532: Configurator 1.1.0 Never Recovers From Save/Reboot

CradyCopter, 2015-12-01:
Latest Configurator r1.1.0 never recovers from *save/reboot* with Naze32 boards. Tried several boards running CF 1.10.0 and 1.9.0 with no difference. Same problem in CLI after a *save* command.
Upper left corner displays a red 'Rebooting' while the main window goes blank and [perpetually] displays an animation saying "Waiting for data..."

As it stands now I have two machines that are now unflyable until the issue is resolved (*unless I switch to Baseflight which would just be so very, very wrong*.)
Is there any way I can downgrade to an earlier, less fancy looking version of the Configurator?

jtp755, 2015-12-01:
Have you tried flashing Betaflight? Ive been using it with no issues on a Naze32 with no issues. Used it on Configurator .66, 1.0, 1.1. Ill try flashing CF here in a bit.

CradyCopter, 2015-12-01:
I did flash one board with BF.  Same problems.

jtp755, 2015-12-01:
You can grab older versions of Configurator from here: https://github.com/cleanflight/cleanflight-configurator/releases

CradyCopter, 2015-12-01:
Cool!  Thanks.

jtp755, 2015-12-01:
When you flash...are you wiping configuration and/or restoring a CLI dump? That will cause problems.

CradyCopter, 2015-12-01:
I have not attempted a restore.  I did try flashing with full erase on one board.
Nothing seems to be working.

jtp755, 2015-12-01:
I would suggest removing the Cleanflight Configurator app from Chrome, restarting Chrome, and reinstall from the Chrome App Store.

CradyCopter, 2015-12-01:
Didn't help.  I can successfully do a 'save' under tabs like PID Tuning but the Configurator get stuck on anything that initiates a reboot.  After that the only way out is to unplug the USB cable.

jtp755, 2015-12-01:
I can reproduce it from the CLI using Configurator 1.1.0 and Betaflight 2.1.4 on my Naze32 Acro.

It actually does initiate the reboot and completes but you cant do anything else in the CLI window. You much change to another tab and then back.

Can you trying clicking on the Save and Reboot button in the bottom right of the Configuration tab? Does the lights on your FC change? What does the log below the CleanFlight logo say?

jtp755, 2015-12-01:
The only place i see an issue is in the CLI window and that works...just does not return to a usable state.

CradyCopter, 2015-12-01:
On a hunch I changed the USB port number and now it's working with one exception: If I use the firmware flasher and attempt to load the firmware online results in sticky reboots.  If I load the firmware locally everything seems to work as expected.
 
I've tried flashing both CF and BF locally with no issues and save with reboots now behave normally with both versions.  Weeeeeeird.

jtp755, 2015-12-01:
I used to have that issue with Android ROMing =)

jtp755, 2015-12-01:
Glad you got it working. I have not run in to the sticky reboot but have not used Load Online in a while either. Im headed in to a conference call but can test after if you want.

CradyCopter, 2015-12-01:
COM ports can be persnickety little bastards.

SteveAmor, 2015-12-02:
Can this be moved to the Configurator repo please.

jtp755, 2015-12-02:
@SteveAmor Is it possible to move the thread or does @CradyCopter have to create a new thread?

@CradyCopter I flashed CF 1.10 using Configurator 1.1.0 using Load Firmware Online with no issues or sticky reboots. Was able to configure everything with the required reboots as well. CLI returning to a usable state was still there though.

CradyCopter, 2015-12-02:
@SteveAmor I cannot find a method to move this so I am closing it.

closed 2015-12-02 by CradyCopter

jtp755, 2015-12-02:
@CradyCopter Just FYI...I opened an issue here: https://github.com/cleanflight/cleanflight-configurator/issues/285