cleanflight/cleanflight #1533: Cleanflight hangs after adjusting PID values through MWOSD

joshuabardwell, 2015-12-01:
I am running MWOSD off of SoftSerial2 at 19.2k. Cleanflight regularly hangs after I adjust PID values through the OSD. The hang happens immediately after the "save and exit" operation. The changes are saved, but the board locks up and becomes completely non-responsive. All LEDs are lit. The board does not arm or respond in any other way until it is power cycled. MWOSD remains functional although it loses communication with CleanFlight. The lockup does not happen every single time, but perhaps once every 3-5 times.

I have searched for the same issue being reported other places, but have not found this exact issue. One similar issue is #1249, however that one has been closed so I'm opening this one separately.

rodestar99, 2015-12-01:
.I have encountered the same problem on two differennt boards . The data gets saved but you have to
unplug to reboot.

eijowk, 2015-12-02:
The combination of MSP and  GPS on Softserial 1 and 2 or vice versa seems to give this problem, also with the cleanflight configurator or Ez GUI groundstation. If GPS is disabled in configurator it works OK. I have had the same behaviour on all versions I have used from about one year ago and on two different Flip32 boards. Same behaviour on 1.11 RC1

I guess that it is stack or buffer overflow? Something is filled faster than it is polled, and then it overflows and some other memory is destroyed.

joshuabardwell, 2015-12-02:
I do not have anything else active on SoftSerial except for MSP to the OSD. That was my main motivation for opening this as a separate issue.

krzysztofmatula, 2015-12-14:
I've recently reported similar issue (#1558). I basically have factory controller setup  with only PPM receiver configured and a single MSP connection on SoftUart. It is used for fetching the telemetry elements using the MSP protocol and causes the controller to hang immediately and always just after boot.

closed 2017-02-25 by hydra