cleanflight/cleanflight #1537: Update "Building in Eclipse"?

conkerkh, 2015-12-02:
Is it possible for someone to update documentation on how to set up Cleanflight to develop in Eclipse, it seems that this Documentation is copied from base flight source and a bit outdated.

ref: https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Eclipse.md

Bengt-M, 2015-12-25:
+1
Not just old. It doesn't work.

CraigJPerry, 2016-01-22:
I followed the steps on http://gnuarmeclipse.github.io/install/ in order. I skipped the jlink step (don't have one). I used the openocd (and qemu) binaries from that projects GitHub downloads.

I'm in windows so even after all that, clean flight wouldn't build, complaining that make failed to spawn another command "CreateProcess(... Failed". To fix that I added msys bin dir to my path in the eclipse environment (i don't want it on my system path). Although cygwin bin would work, it's just looking for some Unix binaries.

I see there's a vagrantfile in the root of the project. Maybe the core devs use that to skip all the setup hassle because that will give a working compiler chain out the box.

Bengt-M, 2016-01-22:
Yepp, same here on my win laptop. It's like it cannot find the path to mkdir. No such path in win.

In a cygwin terminal window it will make fine. I followed the "Building in windows" instruktion and added c++ compiler (needed by unit tests), and set the path to include the arm things (where I used the latest 4.9 version of GCC ARM Embedded).

This is not a good solution because building in cygwin is very slow and clicking in eclipse is much easier than reading file name and line number in the terminal and then locate the issue manually. But anyways.

On my linux desktop it's working as it should with eclipse. One solution I've tried is to run the linux box from the laptop with remote desktop. I have to reconnect sometimes and the touchpad doesn't work perfect but anyway. So far only tried within my own wifi range, but maybe I can tunnel it with putty/ssh some day.

So I guess the "core devs" are using mac or linux.

CraigJPerry, 2016-01-22:
Fwiw Linux and Mac are *far* better developer platforms. I'm only on windows just now due to a quirk of fate but I'd much rather have a decent Unix shell!

All that said, I build in eclipse just now - it does use makefiles not the native eclipse builder but the point is it's all done in eclipse, there's no external shell in my workflow but it sounds like there is in yours? I can share screenshots etc if it helps.

Having moved from eclipse years ago for java and web development I'd forgotten all the keyboard shortcuts and settings I used to need in eclipse!

The JetBrains guys have a new product CLion, I'll see if i can get that working and write it up. It's not a mature ide yet - it's still missing essential features like a debugger that can remote attach, even use of an arm compiler chain will be non trivial but it'll still be worth it just to leverage the JetBrains editor / refactoring / analysis tooling.

Bengt-M, 2016-01-23:
In the post before you said it fails.
How did you fix that?

CraigJPerry, 2016-01-23:
Right click the project folder in eclipse -> properties -> c / c++ build -> environment

Edit the PATH variable and add the bin dir of cygwin or msys at the end of the list. E.g. c:\mingw\msys\1.0\bin or if you have cygwin installed instead then c:\cygwin\bin

Bengt-M, 2016-01-23:
Wow, thank you! That made a big improvement.
Someone should really update the instruction in /docs/develoment.

Now a default make works well and produces a hex file.

Remaining issues with clean and test, however:
```
make test 
cd src/test && C:/Program Files (x86)/GNU ARM Eclipse/Build Tools/2.6-201507152002/bin/make test || true
c:/program files (x86)/gnu arm eclipse/build tools/2.6-201507152002/bin/sh: syntax error: unexpected "("
Makefile:742: recipe for target 'test' failed
make: *** [test] Error 2

make clean 
rm -f ./obj/cleanflight_NAZE.bin ./obj/cleanflight_NAZE.hex rm -rf ./obj/main/NAZE
cd src/test && C:/Program Files (x86)/GNU ARM Eclipse/Build Tools/2.6-201507152002/bin/make clean || true
c:/program files (x86)/gnu arm eclipse/build tools/2.6-201507152002/bin/sh: syntax error: unexpected "("
Makefile:696: recipe for target 'clean' failed
make: *** [clean] Error 2
```
Does this work for you?
It is frustrating because I'm kindof interested in the tests.

CraigJPerry, 2016-01-23:
Yeah make clean works, I haven't had a chance to revisit the tests but mine fail for a different reason. Both your failures are the same thing - it seems to be a quoting issue with the variable substitution which is weird because the makefile is correct.

Back on a computer later will take a look

CraigJPerry, 2016-01-24:
@Bengt-M please test the new eclipse doc in https://github.com/cleanflight/cleanflight/pull/1718 and provide your feedback

gpolic, 2016-02-05:
Been trying to compile Cleanflight and Betaflight for about a week on Windows / Eclipse.
I will give this a try and let you know.

CraigJPerry, 2016-02-05:
Cool thanks @gpolic

gpolic, 2016-02-05:
I will do that when I am on my desktop and have some spare time to uninstall everything and install again as per the instruction.

However in the meantime, I found something else.
If you use the Cloud9 service at http://c9.io you can logon with your Github account and create a new workspace, by cloning your repository. It happens by the click of a button! Already had cleanflight fork on my github
Then I literally use 4 commands as seen on the Ubuntu development file. (for ARM stuff)
"make", and viola !!  there your ELF file...
That was exciting!  You even dont have to install ubuntu or fiddle with your PC or anyhting if you just want to read the code and try stuff.
If you mess everything up there you can delete the workspace and create a new one and you re fresh and good to go!  Its stupid !  I even have some screenshots and some notes down.
It has a Github GUI for noobs like me and some other cool stuff

CraigJPerry, 2016-02-05:
Yeah, there's a vagrant file in the project root, compiling is easy - type vagrant up then make. Getting a usable dev env with hardware debugging etc is harder though :-(

gpolic, 2016-02-05:
Sorry mate I have no luck here. I did the following steps (on Win 7 64bit)

Software Installation - step by step
JDK, I already have 1.8 64bit
Eclipse IDE for C/C++ Developers Windows 64bit  - Luna
gcc-arm-none-eabi-4_9-2015q3-20150921-win32.exe
gnuarmeclipse-build- tools win64 2.6-201507152002-setup.exe
minGW   <---- Link not working
cygwin setup-x86_64.exe
Cygwin <---installation steps are found on "Building in Windows"
stsw-link009.zip  drivers
gnuarmeclipse-openocd-win64 0.10.0.201601101000-dev-setup.exe 
gnuarmeclipse-qemu-win64-2.4.50-201510290935-dev-setup.exe 

Check-out Cleanflight e.t.c. Download project normally
New Project Wizard...  I dont have the ARM GCC in there !  Make sure to make this step before starting the Import: 
--Add a new update site to Eclipse named "GNU ARM Eclipse Plugins" with the URL "http://gnuarmeclipse.sourceforge.net/updates" and install all the features offered <-- This repository is not working
Try some more:
http://gnuarmeclipse.sourceforge.net/updates  <-- not working
http://sourceforge.net/projects/gnuarmeclipse/files/Eclipse/updates/plugins/   <-- not working
Finally I found this page: http://gnuarmeclipse.github.io/plugins/install/  and I drag the Install icon on the Eclipse workspace.  It still fails to install
Then I go to workspace - Menu - Help - Eclipse Market place
search for ARM 
I find the plugin, click install and then --> Repository not found.

The plugin was installed on my previous Eclipse Mars, but it was not compiling correctly (some missing include files)
I will try some more..  Its pretty far from straightforward thought !

gpolic, 2016-02-07:
I went back to Eclipse Mars. Now I can install the ARM plugins, not problem !
So this part is now working:
**--Add a new update site to Eclipse named "GNU ARM Eclipse Plugins" with the URL "http://gnuarmeclipse.sourceforge.net/updates" and install all the features offered
**
I guess Luna version did not work so good.

Installed the GNU ARM, GCC ARM stuff and another GCC compiler from here: http://www.equation.com/servlet/equation.cmd?fa=fortran

One more thing about the instructions.
On the **Short Version** there is a comment:
Import the project using the wizard Existing Code as Makefile Project

If I import the project this way, and try to build, there is an error it cannot find the *.c files and it fails. (also it cannot make the folder /obj/NAZE or something) and it doesnt start to compile.

On the **Long Version**, there is a different way:
Create a new C project in Eclipse and choose ARM Cross Target Application and your ARM toolchain.

Now if I open the project this way, it start to compile normally for .. most of files.
But after a while I receive the following error:

Invoking: Cross ARM C++ Compiler
arm-none-eabi-g++ -mcpu=cortex-m3 -mthumb -Og -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -ffreestanding -fno-move-loop-invariants -Wall -Wextra  -g3 -DDEBUG -DUSE_FULL_ASSERT -DSTM32F10X_MD -DUSE_STDPERIPH_DRIVER -DHSE_VALUE=8000000 -I"../include" -I"../system/include" -I"../system/include/cmsis" -I"../system/include/stm32f1-stdperiph" -std=gnu++11 -fabi-version=0 -fno-exceptions -fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics -MMD -MP -MF"src/test/unit/alignsensor_unittest.d" -MT"src/test/unit/alignsensor_unittest.o" -c -o "src/test/unit/alignsensor_unittest.o" "../src/test/unit/alignsensor_unittest.cc"
../src/test/unit/alignsensor_unittest.cc:23:25: fatal error: common/axis.h: No such file or directory
 #include "common/axis.h"
                         ^
compilation terminated.
src/test/unit/subdir.mk:84: recipe for target 'src/test/unit/alignsensor_unittest.o' failed
make: *** [src/test/unit/alignsensor_unittest.o] Error 1

This is the exact error I was having a few days ago, so I am back to the previous situation.
How can I make sure that the compiler can find all the include files even when they are located in another folder in the project ?

gpolic, 2016-02-07:
Well I got some success finally.
I deleted the GCC and installed Cygwin as per instruction in 
https://github.com/cleanflight/cleanflight/blob/master/docs/development/Building%20in%20Windows.md
now it works, I build some HEX for NAZE and CC3D
There is only one warning !!!!

To summarize at least for the Windows platform, Eclipse Mars works  (Luna I dont think so)
No need to add the GNU ARM and GCC ARM in the system Path as it happens automatically.
I added  c:\cygwin\bin on the path manually.

Then import the project:  Import the project using the wizard Existing Code as Makefile Project

Then go to Eclipse Project/Make Target/Build and add the target names that you need i.e. CC3D and some more if you want.
Then insode Eclipse Project/Make Target/Build select the target and click on button "Build"... it takes some time :-))))

Glad I made it happen finally !

gpolic, 2016-02-08:
And one more important thing:
Before the build, you have to go to File/Properties/C/C++ Build/Build Variables 
there enter a variable. I enter TARGET with values CC3D
Otherwise the build does not start.

There are different options presented in Eclipse menus depending whether:
1. You clicked your cursor inside a text/code file that is opened i.e. main.c
2. You clicked on the Navigator on your project name

The point where you last clicked, is changing the options in the menus

gpolic, 2016-02-08:
Craig, it seems that I got a little carried away with this and forgot to comment on the instruction!
One issue for me is that with Eclipse Luna I could not get the ARM plugins installed.
Although it worked with Eclipse Mars.

One comment I would have is regarding Windows users. More specific: in the Checklist part:
"Windows platform only: Download and Install either Cygwin or MinGW MSYS"
There is a link to download Cygwin at  https://cygwin.com/install.html

I preferred to use the instruction found in "Building in Windows.md" because it is step-by-step instruction telling you exactly what to do. I would link to that one

Overall the instruction looks very detailed and complete.

I didn't try the Git stuff, because I have Github desktop app, and I didn't use the Git plugin for Eclipse.
Also I finally skipped the debug stuff because I do not have a dev board.

Bengt-M, 2016-02-08:
I still say it is too complicated.

I have realised that to build the tests I need a native build which on windows means cygwin (or similar). I have also given up on Eclipse's own way and accepted that I want to use the supplied makefile anyway. After taking those steps there is no need for GNU ARM Eclipse\Build Tools.
(If I wanted to do only cross target the Eclipse way then I should not need cygwin at all. Having both must be wrong.)

I followed "Building in Windows.md" to install cygwin, but added g++.

Now I have no tools in the system path.

No need for JDK, Eclipse is fine with JRE.

Fresh install of eclipse-cpp-mars-1-win32-x86_64
Added http://gnuarmeclipse.sourceforge.net/updates as an update site and installed (inside Eclipse).

Inside Eclipse project build environment: PATH = C:\cygwin64\bin;C:\Program Files (x86)\GNU Tools ARM Embedded\4.9 2015q3\bin

I don't use the "Build xxx" commands. I build with "make target", which has a dedicated view (or right-click on the makefile).
When creating these build targets I set the Build Command to "make" or "make -k"

That's it!

The kinds of builds I have tried are all working fine now (make, make -k test, make clean).
Default make produces a hex file. It gives me a warning (File not found - *.c) that I don't understand but it seems harmless. It gives another warning ("I2C_DEFAULT_TIMEOUT" redefined) that seems to be a valid warning. The hex file is binary identical to that from a cygwin command line make outside Eclipse (where I only get the second warning) so I think it's OK. Right?

The only issue now is that Eclipse is indicating many code problems caused by wrong includes, i.e. Eclipse gets it wrong but the makefile is right. There is probably some setting (maybe for the indexer) that I haven't found yet. That probably means I cannot use the refactoring tools so someday I'll get to it.

And I haven't cared about any debugger. I really can't see the point.

CraigJPerry, 2016-02-09:
Thanks for trying it out guys

nveeser, 2016-02-09:
I had some success with Eclipse on a Mac a while back, thought there were a few things which didn't work well.  (Namely Eclipse gets confused by the all the #ifdefs that get pulled in by different Makefile targets. I was thinking that one could cleanup the Makefile to make things easier for an IDE, but I didn't want to step on anyone's toes.)

The docs *are* a lot of steps to get all the pieces working together, and could use a refresher. Maybe next time I get down to trying to do an install, I can try to document the important stuff, if that helps.

gpolic, 2016-02-11:
It is complicated sure. But as long as there is no installer to fix everything for us we have to manually do this stuff :-o

Going back to scratch because I had some issue with compiling C files (in x86)
I removed everything except the JDK.  Did a CCleaner to clear my registry.
Installed as per instruction steps:
Eclipse Mars, GCC ARM Toolchain, GNU ARM Build Tools, Cygwin (with Windows instruction), GNU ARM Eclipse plugins (did not do the QEMU, OCD STLink e.t.c.) 
One more step is to manually place the Cygwin64 BIN folder in the PATH, because my old C codes did not compile with out it.

I did make on Cleanflight normally from the IDE as before. I dont know how to make the test config
Also did compile an old version of Betaflight this time, that I never been able to compile before !!
It works great!  I can also work with my C & Java stuff without any problems

:+1:

CraigJPerry, 2016-02-11:
Nice one!

torsuneson, 2016-06-09:
There is a easy way to fix building in eclipse without using cygwin. 

I have installed MARS 4.5.2 With CDT, Zylin Embedded CDT Plugin  and GNU ARM Eclipse 
ARM tool chain 5.3 2016q1 (https://launchpad.net/gcc-arm-embedded/)
And eclipse build tools 2.6-201507152002 (64-bit) https://github.com/gnuarmeclipse/windows-build-tools/releases

What I found that I got a message of "Error 2" like above. 

In answer https://sourceforge.net/p/gnuarmeclipse/support-requests/163/#f4a3

> if you installed the GNU ARM Eclipse Windows Build Tools, this package is taylored for managed projects make files, it does not include ANY program that might be required in a custom makefile.
> however, since it uses busybox, **you can try to copy busybox.exe as mkdir.exe** (and possibly other executables).

So I did. And that seem to help, then I added , 'sed' and 'uname' by copying busybox.exe and Now it builds!

So in folder C:\Program Files\GNU ARM Eclipse\Build Tools\2.6-201507152002\bin
Copy and rename busybox.exe three times to: mkdir.exe, sed.exe and uname.exe

[image]

(in the picture there is also a git command, that is not nessesary to make it compile)

Bengt-M, 2016-06-10:
Try building the tests first

torsuneson, 2016-06-10:
Well, I got error. 

It looked like It was invoking the native compiler 'cc' 

So I downloaded MingGW-w64 (https://sourceforge.net/projects/mingw-w64/) 

Added folder of installation to path 
in My case C:\Program Files (x86)\mingw-w64\i686-5.3.0-posix-dwarf-rt_v4-rev0\mingw32\bin

In that folder there was no 'cc' so I copied 'gcc' and renamed it 'cc' 

Now the test run for a while... 

```
Running main() from gtest_main.cc
[==========] Running 3 tests from 1 test case.
[----------] Global test environment set-up.
[----------] 3 tests from FilterUnittest
[ RUN      ] FilterUnittest.TestFilterApplyAverage
[       OK ] FilterUnittest.TestFilterApplyAverage (0 ms)
[ RUN      ] FilterUnittest.TestFilterApplyAverage2
[       OK ] FilterUnittest.TestFilterApplyAverage2 (0 ms)
[ RUN      ] FilterUnittest.TestFilterApplyAveragef
[       OK ] FilterUnittest.TestFilterApplyAveragef (0 ms)
[----------] 3 tests from FilterUnittest (15 ms total)

[----------] Global test environment tear-down
[==========] 3 tests from 1 test case ran. (26 ms total)
[  PASSED  ] 3 tests.
cc -g -pthread -Wall -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/gtest/inc -MMD -MP -std=gnu99 --coverage -DFAST_MATH -Iunit -I../main -D'FLASH_SIZE = 128' -DSTM32F10X_MD -c ../main/config/config_eeprom.c -o ../../obj/test/config/config_eeprom.o
../main/config/config_eeprom.c: In function 'initEEPROM':
../main/config/config_eeprom.c:82:5: error: size of unnamed array is negative
     BUILD_BUG_ON(offsetof(packingTest_t, word) != 1);
     ^
../main/config/config_eeprom.c:83:5: error: size of unnamed array is negative
     BUILD_BUG_ON(sizeof(packingTest_t) != 5);
     ^
Makefile:875: recipe for target '../../obj/test/config/config_eeprom.o' failed
make[1]: *** [../../obj/test/config/config_eeprom.o] Error 1
make[1]: Leaving directory 'c:/Cleanflight/src/test'
Makefile:894: recipe for target 'test' failed
make: *** [test] Error 2
```

But not completly.

Bengt-M, 2016-06-10:
Well, compare with my post from 4 month ago above.
I still say that is the easiest way.

ledvinap, 2016-06-10:
@torsuneson : that is compile-time assertion. It should work in gcc.
Maybe you are using wrong system headers?

can you try 
```
cc -E -g -pthread -Wall -Wextra -ggdb3 -O0 -DUNIT_TEST -isystem ../../lib/test/gtest/inc -MMD -MP -std=gnu99 --coverage -DFAST_MATH -Iunit -I../main -D'FLASH_SIZE = 128' -DSTM32F10X_MD -c ../main/config/config_eeprom.c -o config_eeprom.i
```
in `./src/test/` directory and upload `config_eeprom.i` ?

lit1088, 2016-07-20:
what can i do with NAZE overflowed as following ?
c:/program files/gnu tools arm embedded/4.9 2015q3/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `FLASH' overflowed by 91302 bytes
collect2.exe: error: ld returned 1 exit status
Makefile:983: recipe for target 'obj/main/cleanflight_NAZE.elf' failed
make: *** [obj/main/cleanflight_NAZE.elf] Error 1

Bengt-M, 2016-07-20:
That would mean the code is too big for the target. Maybe you added calls to functions not used before. I had the same issue after adding a call to tan() and resolved it by changing to sin_approx()/cos_approx()

ledvinap, 2016-07-20:
You must enable at least some optimizations on 128k targets to fit in flash. `-O1` may be enough and it doesn't interfere with debugging that much

SandroGrzicic, 2016-10-24:
@CraigJPerry can you please do a quick writeup on how to get started in CLion if you have some time? I have already set up my local development toolchain and I can build the project using make and have even flashed such a locally-built cleanflight version onto my FC, but I am completely lost in CLion, even though I have been working with IDEA/Webstorm for years now (Java/Scala). The error mentions something about   no CMakeLists.txt file being available, which I presume has something to do with the build system being CMake and not using the standard makefile in CLion:

https://www.dropbox.com/s/7wn32e0eupii9b0/2016-10-24%2022_20_55-.png?dl=0

Thanks for any help.

conkerkh, 2016-10-24:
@SandroGrzicic to be honest, the easiest way would be to simply compile everything in Terminal manually and code in CLion.

SandroGrzicic, 2016-10-24:
Actually, that's a great point. In my Scala projects, I do all my compilation in a sbt shell anyway (other than running tests, which can of course also be run within the sbt shell). I suppose it's the same thing here.

Thanks!

closed 2017-02-25 by hydra