cleanflight/cleanflight #1539: Hott Telemetry doesn´t work in Release1.11.0RC1 and 1.10.0 on Flip32

Flugsicherung, 2015-12-03:
Hallo, 
I have a Flip32 10DOF with all sensors an I have a sonar installed. In Release 1.09 Hott Telemetry is working but on 1.10.0 and 1.11.0RC1 Telemetry dont work !!!!! Hot Telemetry is on Softserial1 and Sonar ist on Softserial2. Please check this problem.
PS: On CC3D I test the Hott Telemetry on 1.09.0 and on 1.10.0, both is working!!!

droidparanoid, 2015-12-20:
Hi!
I can report the same issue like Flugsicherung.  Here are some more detailed information about this issue.

My setup is slightly different:
- Naze32 acro rev 5 (no extra features, just the plain flight controller)
- Graupner GR-12L receiver and MX12 radio
- the GR-12L has "sensor" as input at channel 5 selected (in the radio menu)
- the GR-12L is connected at USART2 via SUMD
- Hott telemetry at Softserial1

I have fully erased the board and flashed it with Cleanflight release 1.9 and 1.11 to test this issue. At 1.9 Hott telemetry works perfectly, at 1.11 (and also 1.10) no Electric Air Modules is listed in the sensor tab in the radio menu.
For both tests I have applied the exact same values (only rx_serial, softserial, telemetry and oneshot are changed from the default values).
The receiver and MX12 has the newest firmware, so I can confirm that it is NOT a problem of the Hott update with the auto sensor detection.

Attached are the settings for release 1.9 and 1.11.
[1.9.0.txt](https://github.com/cleanflight/cleanflight/files/67865/1.9.0.txt)
[1.11.0.txt](https://github.com/cleanflight/cleanflight/files/67866/1.11.0.txt)

I would very appriciate if someone would find a solution for us. Unfortunately I am not very familiar with coding and cannot check the differences in the cleanflight code of the two releases.

Thanks a lot!

jo930, 2015-12-20:
I have version 1.11 on my Naze32 acro rev 6 and Hott-Telemetry is working without issues for me.
Make sure to select "HoTT" and "19200" in the Ports-Tab in line SoftSerial1 and column Telemetry.

Flugsicherung, 2015-12-21:
Hi jo930,
jes we make the right settings on the Ports-Tab. In version 1.9 Hott Telemetry works fine!!!
have you make a full erase with the flashing of version 1.11? How did you make the flash update?
Flash local with the hex File for the Naze or did you make in online menu?
Do you use Hard Serial Ports? I use all 4 Serial Ports and I thing here is the Problem.

droidparanoid, 2015-12-21:
Hi jo930 and Flugsicherung,

yes, Hott is selected at Softserial1 in the ports-tab, as well as telemetry activated. All settings of the cleanflight configurator are identical in release 1.9 and 1.11. In addition, I also tested all available Baud rates for Hott at softserial1 in the ports-tab, none works in 1.11 ("Auto" works perfectly in 1.9).

I flashed both versions over the online menu with the option "Full Chip Erase".

At the Naze32 rev5 the receiver is connected in this way:
GR-12L (channel 6) <---> USART2 RX (board identifier 4)

and for telemetry:
GR-12L (channel 5) <---> Softserial1 RX (board identifier 5)
GR-12L (channel 5) <---> Diode <---> Softserial1 TX (board identifier 6)

As I wrote, no extra features are connected to the board, only the ESCs + motors. The telemetry wiring is done exactly as described in the docs, and works perfectly with release 1.9.

Flugsicherung, 2015-12-21:
Ok,
I thing the Problem is in Version 1.10 and 1.11 the changes for the Revision 6 of the Naze32.
Boards with Rev.5 have the Problem, Rev.6 Boards works perfect........

Pierre-A, 2015-12-21:
The issue was related to bad synch as described in issue #1413 and solved in PR #1419 which has been merged in last official release.

So I just flashed a Naze32-rev5 (associated to a GR12L) with Cleanflight v1.11.0 and with same configuration than **droidparanoid** : Hott telemetry (EAM) works fine. And works fine too with the development version.
So I don't understand what happens. Can you describe a bit more ?

Flugsicherung, 2015-12-21:
Hello Pierre-A,
I flash 1.11.0 with full erase on my Flip 32 DOF 10 Board again, but Hott doesn´t work.
Here ist the Dump of the Config
[CLI_Flip32_V1.11.0.txt](https://github.com/cleanflight/cleanflight/files/68607/CLI_Flip32_V1.11.0.txt)
I use all Sensors and GPS and Sonar. Same config on Release 1.9.0 works perfect but on 1.10.0 and 1.11.0 I can´t get it running. I thing the changes off Revision 6 of the Board brings the Problem to Revision 5 Board Users. Tell me please how I can help you to find the bug.

Pierre-A, 2015-12-21:
**@Flugsicherung**, thanks for replying.
Looking at your dump, I see that you selected `feature RX-PPM` (so can't be a GR12L) and you set `serialrx_provider = SPEK1024`(so not a Graupner). You can't have Hott telemetry for sure.
What is your RX ?

Flugsicherung, 2015-12-21:
Here are some Waveforms of the Telemetry Signals in Version 1.9.0 and 1.11.0 both on Soft Serial 1 with same Parameters.

[image]
[image]

Flugsicherung, 2015-12-21:
I have a Graupner GR-24  and a MX22 with M-G2 Module, my MZ-24 Transmitter is in repair.....

droidparanoid, 2015-12-21:
Hmm I dont know what could be different. Here my steps in detail:
1. Flashed 1.11.0 with "full chip erase"
2. Port tab: activate Serial RX at UART2
3. Config tab: activate RX_Serial and SUMD; and feature Softserial, Telemetry and Oneshot
4. back to Port tab: activate Hott at Softserial1

Thats all. No further modifications. The copter is wired as described above, only:
- Naze32 acro rev5
- GR-12L with signal and telemetry as described above
- single VBAT wire from main power
- ESCs and motors

No further setup. I have tested:
1. first radio on, than power on copter
2. first power on copter, than radio
3. step 1 but with UART1 connected via usb at cleanflight configurator
4. step 2 but with UART1 connected via usb at cleanflight configurator
5. I activated the setting "set telemetry_inversion = ON"

Here some infos for the radio/receiver:
- MX12 is firmware 1.799
- GR-12L is firmware V3.67
- the GR-12L has following menu options: "CH OUT TYPE: SUMDOF06" and "5CH FUNCTION: SENSOR"
- I have tested the function "PERIOD" with 10 and 20msec

What else we could try? And as can be seen in my dump file at the beginning there is feature RX_SERIAL and set serialrx_provider = SUMD correct.

Flugsicherung, 2015-12-21:
The SPEK1024 Parameter is not for PPM, only for Serial Receivers and shoud not have any effect when using PPM with SUMO

Pierre-A, 2015-12-21:
**@Flugsicherung** , oh yes, my bad, I had GR12L in mind... the first wave form is as expected but the second is corrupted. No difference in the wiring on your model ? I have a GR24 embedded on another model and should try with it (needs to take it out first..).
**@droidparanoid** , I have the same configuration with MX20 (v 1.200). I'm gone plug my scope again on the telemetry pin.

Flugsicherung, 2015-12-21:
Yes I tested it on the same drone, so the wireing is exactly the same for both screenshots  ;) 
I am not so got with programming but in Hardware I am a professional. If you make any changes, my son can build the hex file.

Pierre-A, 2015-12-21:
OK OK, thanks, I'll let you know..

Pierre-A, 2015-12-21:
**@Flugsicherung** , **@droidparanoid** ,  I just scoped the telemetry pin with the current development version : Hott signal and values are fine (GR12L + MX20).

Would you mind testing this and compare with the v1.11.0 on your config ? (no need flying).
If so , please PM me I'll send you the Naze.hex.

Flugsicherung, 2015-12-21:
Ok I send you a PM

Viele Grüße
Tombhunter

droidparanoid, 2015-12-21:
Sended you PM too :-)

droidparanoid, 2015-12-21:
Hi Pierre,
I've tested your dev version, unfortunately the same behavior like with 1.11. I also would assume that there is a bug due to the support for naze rev6 now, as Flugsicherung wrote...

I don't know if it is of interest, I also tested the newest betaflight from boris without working hott telemetry.

Is there anything else I could test on my setup? Maybe there is a problem between GR12L and MX12, but why is it working on 1.9.0 then?

droidparanoid, 2015-12-22:
Something I found out now, at my setup I have installed the GR-12 and not the GR-12L. Sorry for the confusion, I have both receivers and I confused them.
Unfortunately my other GR-12L is not here (Iam on holidays), so I can test him only in some weeks when Iam back home. Maybe there are differences between both typs with effect on the hott telemetry...

Flugsicherung, 2015-12-22:
It is definitely not a Problem on the Graupner devices. I have Gr-24 and Gr-12L Recievers and I use Transmitters MZ-24 and a MX22 with M-G2 Module. Everything works perfect in CF 1.9.0 but nothing is working on CF 1.10.0 and 1.11.0 so the Problem is in the changes for supporting the Naze32 Rev.6

Pierre-A, 2015-12-22:
Not related to GR12 L or not , (this has been tested a while ago and revealed no real difference on the telemetry side).
I spent some time this morning looking at the commits history with no luck yet.. have to dig a bit more..

Flugsicherung, 2015-12-23:
Hi Pierre-A,
is your Board a Naze32 Rev.5 arco or have you all Sensors onboard? The Flip32 Board is with all Sensors but the Rev. is <5 because the 8Mhz Clock??
@droidparanoid 
your Board is a Naze32 or a Flip32 acro without any Sensors right?? Can you please expain in detail?

droidparanoid, 2015-12-23:
Yes, I have the Naze32 acro Rev.5, only with gyro and acc, no extra sensors.

Flugsicherung, 2015-12-23:
Can you send me a photo of your controller?
If that is a Naze 32 Rev.5 Pierre have the same and no problems ???
I can't understand that.

droidparanoid, 2015-12-23:
I cannot make a direct foto from top because I fixed the top plate of the copter with cable ties and I don't want to remove everything. But I have the exact model like illustrated here (AfroFlight32, rev5; the blue one):
https://www.multirotorparts.nl/electronics/flightcontrollers/naze32-acro-flightcontroller-rev5.html

Flugsicherung, 2015-12-23:
Ok it is  a Naze. But why doesn't CF 1.11.0 runs on your controller and Pierre have the same  one and he can run it?
I am total confused at the moment.....

Pierre-A, 2015-12-23:
OK guys, sorry I was a bit silent today but I did a lot of tests trying to figure out what happens to Hott telemetry in the code, using different configurations and with scope on telemetry pin  :
    - RX : GR12L and GR16
    - boards : Naze32rev5 and SPRACINGF3
    - FW :  v1.9 , v1.10 , and current Cleanflight master (commit #791144e9).

First of all, as you already observed it, the issue begins as soon as v1.10 with Naze32 and is in current dev version.
**pict 1** : commit #79144e9 on Naze32 with GR12L :  the telemetry works but the signal is already affected (-0.7 V on high level).
[image]
**pict 2** : commit #79144e9 on Naze32 with GR16 : the telemetry does not work due to a high voltage drop (-2.2 V)
[image]

This can be compared to the case of SPRF3 with both RX (pictures 3 and 4) where no issue is encountered :
**pict 3** : commit #79144e9 on SPRF3 with GR12L : signal unaffected, telemetry is fine.
[image]
**pict 4** : commit #79144e9 on SPRF3 with GR16 : a small drop with no incidence on telemetry.
[image]

So, conclusion for tonight : 
    - issue specific to Naze32, and not only specific to Flip32,
    - telemetry with GR12L works on Naze32 but it's a fake,
    - GR16 is much more sensitive than GR12L (probably due to a different input impedance on telemetry side),
    - the issue is probably not related to the code supporting the Naze rev6 (merged in v1.11),
    - have to dig a bit more in the source...

Your comments are welcome !

**@hydra** , any other idea ?

Flugsicherung, 2015-12-23:
Great work, I am happy that you get the same issue.  ;))
Thanks for the time you spend to this issue, I think the Problem is in the right Hands.
It might helpful to see in CLI "Status" more details like what Board Type and Hardware Revision the Firmware is detecting. I guess here could be the problem. If you can change the Status output I can make a new test with my controller.

droidparanoid, 2015-12-24:
Thanks Pierre for testing this out! I hope we get to the core of the problem now :-) and Merry Christmas to all!

Pierre-A, 2015-12-24:
hopefully in the tree.. and MERRY CHRISTMAS you all too !!

Flugsicherung, 2015-12-24:
Today I get my MZ-24 Transmitter back from the Repair :)
Now I'm ready to make tests again.
MERRY CHRISTMAS and many Gifts. :))

Pierre-A, 2015-12-24:
After a long day digging in the code..
it appears that the level of the hott signal is affected when defining the serial output port (call to `serialOutputPortConfig (softSerial->txTimerHardware`  in drivers/serial-softserial.c/openSoftSerial, line 225).
The level is pulled to ground and it looks like the port is overwritten but I did not find where.
This made me think of an impedance adaptation so I tried.. pulling the signal to +5V with a 22Kohm resistor and ooops... the signal is fine and the hott telemetry works like a charm (picture 5 below).
Can you try that on your config ?
That said, it works but I have not yet find out the origin of the different response between v1.9 and v1.10.

**pict 5** : commit #79144e9 on Naze32 with GR16 (same as pict 2 above) but pullup 22K resistor : no more drop on telemetry.
[image]

Flugsicherung, 2015-12-25:
Wow, that is a perfect Christmas Gift. :-)))
I flash Version 1.11.0 and build in the 22k Resistor on the Telemetry wire.
The Signal looks like yours......
Hott Signal with Firmware 1.11. 0 and 22k Ohm Resistor
[hott signal with 1 11 0 and 22k ohm](https://cloud.githubusercontent.com/assets/16131122/12001736/1f81e91c-aaef-11e5-8507-14c4c1f203a6.png)
This is perfect !!!! I tested this on GR-24 Receiver and MZ-24 Transmitter.
Now I waiting fa 3D-Fix on the GPS to test last things.   ;-)
Thanks very much for the great work

Flugsicherung, 2015-12-25:
Hello Pierre-A,
one little Problem I found on the GPS display....
In the Configurator GPS shows 131 Meters of Altitude, OSD Display shows the same.
[image]
But on the Transmitter the Display shows only 13 Meters, thats factor 10 to low!!!!
[image]
On the MX22 Tranmitter the Smart-Box shows the same issue so I think it is a little Bug in the Hott Protocol. All other works perfect !!! I am very happy.

Pierre-A, 2015-12-25:
**@Flugsicherung** : OK, fixed it. As I don't use GPS, I sent you a Naze.hex, can you try it ? If it works, I'll open a PR for correction.
Plz, try not to mix the topics.. :wink:

droidparanoid, 2015-12-25:
Hi Pierre!
Thanks for the great work!
Iam not really an expert of electronics, but as I understood the resistor in the telemetry wire is only a workaround for the bad signal, and the real issue has to be fixed in the code, correct? I have no resistor here at the moment and cannot test this for now.

Pierre-A, 2015-12-25:
**@droidparanoid**, yes you're right. This workaround is needed to fix the input level and avoid a conflict on the telemetry line. Have to fix it in the code for the input line.

Flugsicherung, 2015-12-25:
Hi Pierre,
great, GPS Altitude is shown perfect. Confugurator and Transmitter displays the same value.
You can make a new topic for the issue and close it. :-)))
Thanks for you help. I am very happy, its very windy in the next days so I can´t test it outside but inside all work right and I think outside its work too.

Pierre-A, 2015-12-25:
OK guys, no more external resistor !
It was effectively due to a bad impedance adaptation; the timer input mode had to be fixed to pull-up. 
I send you a .hex for testing. 
I still have to look at possible consequences on other items (as ledstrip..).

Flugsicherung, 2015-12-25:
Ok, I solder the Resistor on the Board, close cover and screwed all together. So testing without Resistor is many work.
@droidparanoid: Can you test it please?
I go outside and risk a flight ;)

Flugsicherung, 2015-12-25:
Ok, I can´t fly with the GPS Altitude correcting Hex file. Motor spinning wrong on Gas....
I flash the last Version (without Resistor) and test again.

Flugsicherung, 2015-12-25:
Hi Pierre-A,
I make a test with your last Version with Hott Fix. Motors spin correct and I take a fly outside.
Hott Telemetry funktion is correct  (Altitude too). Out of the Box with the standard MultiWii Parameters Baro and Sonar won´t work correct but without calibration I think its normal. So I must do more Tests outside. Yeah  the Copter on my Acccount Photo is flying ;-)))

Pierre-A, 2015-12-25:
Great, glad it works and thanks for testing !

Flugsicherung, 2015-12-25:
Thank you for debuging the Problem ;-)))

droidparanoid, 2015-12-25:
Hey guys!
The new hott fix is working perfectly for me! :-) I have valid telemetry data and the copter behaves normal (on ground so far). I will fly tomorrow maybe.
Since this is a development release, can I fly with it regularly until the next stable release or are there any surpises I should be aware of?

Thanks Pierre for the great and fast work on this issue! :+1:  :-D

Pierre-A, 2015-12-25:
No worry, you can fly with it.  It's only a little fix based on current dev master.
If you ever notice any other anomaly, feel free to open a new issue.

Pierre-A, 2015-12-28:
OK guys, much much easier when one knows what to look for...
I found the line in the code where this has been changed and the related commit merged in Cleanflight since v1.10.
**@hydra**, I sent you a mail, thanks for your comment.

Pierre-A, 2015-12-29:
I opened PR #1637 which should be merged quite soon.
This issue can be closed.

Flugsicherung, 2015-12-29:
I close this issue, thanks.  ;-)

closed 2015-12-29 by Flugsicherung

hydra, 2015-12-30:
Merged the related commits from the PRs.