cleanflight/cleanflight #1542: Modes/aux channels are not working on naze32 with spektrum RX

Austinthemighty, 2015-12-04:
Just flashed the latest version of cleanflight to my naze32 v6 with spektrum RX and TX, went into the receiver tab changed the channel map to TAER1234 and then went into modes and set my aux 2 switch to arm and my aux 1 to angle, and horizon mode, saved, turned on the TX and my new versacopter from flitetest which is running a naze32 v6 and it would not receive a signal from the aux channels, only way to arm it is throttle low and right yaw

jtp755, 2015-12-04:
Double-check your receiver setting on the Configuration page as well as this in the CLI: serialrx_provider

Austinthemighty, 2015-12-06:
i already did that, 3-4 times

SteveAmor, 2015-12-09:
Assuming you are using PWM, does the Aux channel work if configured as AETR?

closed 2017-02-25 by hydra