cleanflight/cleanflight #1547: rxrange has no effect on Channel 1

papaiatis, 2015-12-06:
I have flashed CleanFlight 1.10.0 on CC3D and trying to setup PWM RX with Turnigy 9x. 
The TX has range from 1036 to 1860. 
I omitted the rxrange command for all channels eg: "rxrange <ch num> 1036 1860". All channels are now properly configured except Channel 1 (Roll). It seems "rxrange 1 1036 1860" has no effect at all.
Any idea?

papaiatis, 2015-12-06:
So I've just figured out that channels are numbered from 0 and not 1. 
Please mention it in the documentation!

joshuabardwell, 2015-12-06:
>Please mention it in the documentation!

Please submit a pull request! Editing documentation is one of the easiest ways for non-programmers to contribute to this project. Learning to edit with Markdown is not too hard and you can make the edits right here in the web GUI, no need to clone the repo locally.

closed 2015-12-06 by papaiatis