cleanflight/cleanflight #1549: ESC Programing mode enhanchment

Austinthemighty, 2015-12-06:
It would be nice if there was a way to program your esc from your flight controler, like if there was a tab that you went to and you could use your arrow keys to control the values of the output. it would be something similar to testing the motors where you set the output speed of the motors.
something simmilar to this.
https://www.youtube.com/embed/AjLJZlG2TnU
and this
https://www.youtube.com/watch?v=E6epG5E8gAQ

joshuabardwell, 2015-12-06:
You mean like this?

https://www.youtube.com/watch?v=YWLk4qcQcvw

dkisselev, 2015-12-06:
I think he's asking to add a full BlHeli configuration UI into CF configurator?

BLHeli and Cleanflight are totally independent projects so BLHeli might add/change features or commands to their serial protocol at any point. I don't think it's worth the Cleanflight team's time to track these changes and constantly update the cleanflight configurator to make sure it can still configure BlHeli devices, so the above video is the best way to do it anyway.

joshuabardwell, 2015-12-06:
I agree. Passthrough allows the projects to remain in separate development cycles, which is ideal.

Austinthemighty, 2015-12-07:
Something like where I plug in the battery to the quad and then I hit the up and down arrow keys to simulate the throttle from my transmiter, it is really annoying programing each individual esc one by one.

Also I was not able to watch the video at the time of writing this as it is currently block at my location.

dkisselev, 2015-12-07:
Why not just use the motor output tab to do that like you linked above? It does exactly that (simulates transmitter outputs to the esc's).

joshuabardwell, 2015-12-07:
Ahh, yeah. The Motors tab totally already does this.

ISSUE CLOSED!

Austinthemighty, 2015-12-08:
I had a hard time doing that way, even when using a mouse or a touchscreen computer, how about a switch in the motors tab that you can slide and it will jump all the way to the top or to the bottom when you press the up and down arrows

joshuabardwell, 2015-12-08:
Left-click on the motor slider, then use up/down arrows to make fine adjustments or Page Up/Page Down to make broad adjustments.

ISSUE CLOSED! (LOL)

fftunes, 2016-03-19:
Or just click the very top and very bottom of the motor slider. It will jump to endpoints.

Austinthemighty, 2016-03-19:
that was what i was doing until now, now i am pressing home and end.

closed 2017-02-25 by hydra