cleanflight/cleanflight #1550: G-Tune in v1.11.0-RC1 fw / v1.1.0 configurator

jaffadog, 2015-12-06:
I am experimenting with the v1.11.0-RC1 firmware, which lists G-tune as one of the changes introduced.  Configurator v1.1.0 is prescribed for this firware version, but I'm not seeing any option to eable G-Tune in the "Modes" tab.  I tried to assign it int he CLI with "aux 2 26 0 1600 2100", but now in the "Modes" tab it claims "Failsave" is enabled by that aux button.

paulwas, 2015-12-08:
I too am experimenting with v1.11.0-RC1 on a Naze32 FC with Baro & Mag but can't find an option to Enable G-Tune. I'm also using 1.1.0 the CFG. It's not listed on the Modes tab in the GUI and I can't find any mention of it or any of the "gtune_.." parameters in the CLI. Is there something else we need to do?

chasers, 2015-12-08:
It's in the inflight adjustments tab ... seems to be working pretty well for me at least.

jaffadog, 2015-12-08:
g-tune is in the "adjustments" tab?  I'm not seeing it there either (in CFG 1.1.0)...

paulwas, 2015-12-08:
Thanks for the tip, I hadn't looked there. Unfortunately, my adjustments list (see .jpg) doesn't show anything for G-Tune. Is there anything else I need to do to enable it? I've verified that I do actually have the right FC flashed (Flight controller info, identifier: CLFL, version: 1.11.0) and I'm running the right CFG (OS: Windows, Chrome: 47.0.2526.73, Configurator: 1.1.0). I've also looked thru all the CLI variables and don't see anything having to do with gtune. I'm guessing there's just something dumb I'm missing..

[image]

chasers, 2015-12-09:
Oh derp I think I meant the receiver tab, wherever you set your ranges for
your flight mode. It's where Autotune used to be.

Here's a good video: https://youtu.be/_16hyAnJJTY

chasers, 2015-12-09:
Yeah what I really meant was the Modes tab!
[image]

paulwas, 2015-12-09:
Thanks for your help and the video link. That is what I thought you meant but GTUNE isn't showing up for me. I see you're on a MAC and I'm on Windows so that might be it. Otherwise there may be something else I need to do to get it to show up. I'll keep digging to see if I can figure it out..

[image]

jaffadog, 2015-12-09:
Chasers, can you confirm what firmware version you are running?

paulwas, 2015-12-09:
Sure. It's 1.11.0, the RC version. And the CFG version is 1.1.0. I've flashed it several times back and forth from 1.10 and 1.11 and re-configured it manually each time. And my FC is a Naze32 with Baro and Mag (10DOF). Don't know how much this matters but I don't have a receiver connected to it. So far I'm just testing the FC stand-alone. I have tried powering it with a BEC before plugging the USB port in but that made no difference.

chasers, 2015-12-10:
[image]

jaffadog, 2015-12-10:
Humm... @chasers has a SRF3 board, whereas @paulwas and i have a AFNA.  The problem seems to be that GTUNE is not enabled for the AFNA:
 https://github.com/cleanflight/cleanflight/blob/master/src/main/target/NAZE/target.h

jaffadog, 2015-12-10:
https://github.com/cleanflight/cleanflight/pull/1557

flounderscore, 2015-12-13:
Same issue with RMRC Dodo. Pull request #1566

GeertMeyer, 2015-12-14:
Same issue With Skyline32

hydra, 2015-12-14:
some changes have been made which allow gtune to fit on the naze board now.  merged.

closed 2015-12-14 by hydra