cleanflight/cleanflight #1552: v1.11.0-RC1 - "set align_mag = 8" responds with "invalid value"

jaffadog, 2015-12-06:
align_mag used to take integer values 0..8.  I see the new default value is "DEFAULT", but I have not seen any documentation of what the new permitted values are.

stronnag, 2015-12-06:
CLI numeric values are somewhat replaced by enumerations. Unfortunately, the documentation is rather behind the implement. See https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_cli.c#L309 for enumerations for alignment, indexed from 0.

This is  at least alluded to in the release notes "CLI usabiity improvements." (sic).

jaffadog, 2015-12-06:
ok found valid values in serial_cli.c.  just need to update the docuemntation.

static const char * const lookupTableAlignment[] = {
    "DEFAULT",
    "CW0",
    "CW90",
    "CW180",
    "CW270",
    "CW0FLIP",
    "CW90FLIP",
    "CW180FLIP",
    "CW270FLIP"
};

closed 2015-12-06 by jaffadog

Flugsicherung, 2016-01-15:
Please update the Documentation for CLI in the Master.... I had the same Problem and I search a long time and than I found this....
Many Thanks

skaman82, 2016-01-15:
It is already up to date: See https://github.com/cleanflight/cleanflight/blob/master/docs/Cli.md

Flugsicherung, 2016-01-15:
Sorry but if I open your Link I get the wrong settings like this "DEFAULT, CW0_DEG, CW90_DEG, CW180_DEG, CW270_DEG, CW0_DEG_FLIP, CW90_DEG_FLIP, CW180_DEG_FLIP, CW270_DEG_FLIP." and this false !!!

skaman82, 2016-01-15:
oh, sorry you are right. I make a PR for that one.

Flugsicherung, 2016-01-15:
Great, thanks.

skaman82, 2016-01-16:
@Flugsicherung Done in: https://github.com/cleanflight/cleanflight/pull/1683