cleanflight/cleanflight #1554: A fantastic addition to the CleanFlight Firmware: OpenAeroVTOL V1.3

poulbran, 2015-12-07:
The Progress made in the VTOL world has been incredible, but the hardware is lacking.
NASA used a KK2 board for the Greased Lightning Project: [CleanFlight could do this]( https://www.youtube.com/watch?v=kXql26sF5uc)
Another advanced project like an F35 also used the KK2, but required additional hardware like the pixhawk to stabilize in horizontal flight: [CleanFlight could also do this](https://www.youtube.com/watch?v=aldZEl_W7VE)

The clear evolution for this type of application is to move to a 32-bit flight controller. 

Cleanflight would be perfect for this because it already has multirotor and plane flight stabilization.
The only addition needed is a transitional mixer. If an add-on is possible that may be the easiest. 

Porting the code directly to CF may also be a possibility.
 Unfortunately, I lack the skills necessary for either of these. 
(Just now taking CS1301 in college)

If there is anyone at all willing to tackle this project, you are sure to see positive results.

[Link to the RCGroups Thread](http://www.rcgroups.com/forums/showthread.php?t=1972686&highlight=NASA&page=144)

Elmeerkat, 2015-12-07:
I'd be very interested in this so I could use the afromini on smaller VTOLs

Linjieqiang, 2015-12-08:
Sounds great！

taileron, 2015-12-14:
naze 32 with CF 1.10 works great for vtol - the only thing that is currently missing are mixers per profile.
One Mixer for tilt of 0 .... 35 degrees the second for around 45 degree and one above up to 90 degree (forward flight) then yaw changes completely to roll and reverses its function. Roll into yaw.
The control and stabilization axis change and the mixers should adapt this.    
Each separate vtol flight situation can be perfectly solved with the mixers of CF 1.11   
Elevons and tilt servos currently are done by an orange RX3S such a receiver also can inject a sbus signal for the CF controller. 

Better would be a second CF controller using the same sbus to stabilize the vtol below stalling angle along the whole transition.

poulbran, 2015-12-15:
So with the built in mixers will already do the job. 

This is wonderful!

Thanks so much for the info!

closed 2015-12-15 by poulbran

taileron, 2015-12-15:
the mixers only work when they are per profile - the current versions only provide one mixer (mmix/smix) for all profiles.
Inflight only rate profiles are remoteable.
It is necessary to have access to change inflight the whole 3 profiles inclusive their mixers / PIDs inflight with an aux channel.