cleanflight/cleanflight #1555: Support for SRF10 sonar rangefinder implemented

martinbudden, 2015-12-08:
The SRF10 rangefinder is smaller than the HC-SR04. It has a range of 6m and is an 5V I2C bus device, which means it needs a logic level converter (about £2) to connect to a typical flight controller. It does, however, cost more than the HC-SR04 (about £25 to £30).

Anyway, I've been working on code to get it going in Cleanflight.

The code is here: https://github.com/martinbudden/cleanflight/tree/sonar_srf10

The code is not quite ready for a pull request.

It working, with two provisos:
i) it's hardcoded to select the SRF10, I haven't got the detection code working yet
ii) the initialisation parameters (i.e. gain and maxrange) of the SRF10 need to be tuned to give best results - I've only tried it inside so far.

Anyone is welcome to try this out, feedback is welcome.

Getting this going also involved cleaning up the sonar sensor and device driver code so that responsibilities of the driver and sensor code are more clearly defined. The upshot of this is that adding new rangefinding sensors should be relatively straightforward. I've already made a first cut at a Pulsed Light LIDAR Lite driver - but I don't have the hardware to test it yet. I'm also not likely to have too much time to work on that until after Christmas.

Here's a photo of the SRF10 hooked up to a CC3D via the FlexPort I2C bus and a logic level converter.

[image]

skaman82, 2015-12-08:
This is great, maybe you could look at the sonar altitude bug also: If sonar is attached and IN RANGE the altitude is displayed correctly in OSD. When it gets out of range the displayed altitude is 0 instead of displaying the current baro altitude.

skaman82, 2015-12-08:
It would be also great if SRF02 would be supported too - its also small and not so pricy like the SRF10

martinbudden, 2015-12-08:
According to this:

http://www.robot-electronics.co.uk/htm/arduino_examples.htm#SRF02,%20SRF08,%20SRF10,%20SRF235

"The SRF02, SRF08, SRF10 and SRF235 all use the same I2C interface. The basic ranging commands are the same..."

So the code should work for the SRF02. (I haven't got one of these, so I haven't tried it.)

If you try it, let me know how you get on.

See also: 
http://www.robotshop.com/media/files/pdf/ultrasonic-rangefinder-comparison.pdf
for a comparison of rangefinders.

Here's a FAQ about the SRF series of sonar rangefinders.
http://www.robot-electronics.co.uk/htm/sonar_faq.htm

martinbudden, 2015-12-08:
The SRF10 detection code is now working. It is a bit of a hack and not yet incorporated into the detection framework.

MJ666, 2015-12-10:
I was testing SRF02 about 2 years ago. For me the sensor we sending erroneous values near the ground (in short distance). I implemented support for the Harakiri firmware at that time but finally decided the for an Maxbotics MB1242 I2C sensor. The behavior of the SRF02 was not really predictable. not sure if the SRF10 is behaving better here. Since the Maxbotics sensor is only working with 100kHz I2C bus speed so I also implemented an dynamic speed switching of the I2C bus. Today with using short loop times and the general slow read of gyro data via I2C is may be not the best solution anymore. My large quad was running standard looptime of 3500 at that time. Here is the link to the code:
https://github.com/Crashpilot1000/TestCode3/blob/master/src/drv_sonar.c
the are also some comments regarding the I2C sonar here:
https://github.com/Crashpilot1000/TestCode3/blob/master/src/config.c#L319
This also have some statements why an logic level converter is not really needed even if the sensor is running with 5V. More info can also be found in the forum thread (sorry but its in German):
http://fpv-treff.de/viewtopic.php?f=18&t=1368&start=3320#p58903

martinbudden, 2016-02-08:
Now working on SciSky flight controller. Here's an SRF10 mounted on a picnic quads 120mm frame and connected to a SciSky flight controller:

[image]

closed 2017-02-25 by hydra