cleanflight/cleanflight #1556: No input signal from ch5 with pwm and vbat disabled

McJohn115, 2015-12-09:
Hi Guys,
im using a CC3D with CF 1.10.0, a Spektrum AR610 RX and TAER settings, all channels works fine but i cant get a input on CH5 (AUX1 - 3 way switch for flightmodes).
I thought it was fixed #612 but it seems not.

Any help for me?

SteveAmor, 2015-12-09:
Duplicate of #1542?  Is it ok if you configure as AETR?

McJohn115, 2015-12-09:
No, all channels interchanged but no signal on ch  5.

McJohn115, 2015-12-10:
Any other ideas?

SteveAmor, 2015-12-10:
Does V1.9.0 work as expected?

McJohn115, 2015-12-10:
No, i tried this 2 hours ago, same problem, and same with Betaflight.
With Openflight and Betaflight all channels works as they should.

McJohn115, 2015-12-11:
Are there any settings needed so i can see any movement on the AUX-Bar under the Reciever Tab in the CF Configurator?
I tried do set the Servoadjust on TX lower and higher but nothing happens.

SteveAmor, 2016-01-08:
@McJohn115   this is a support problem that should be worked through in the forums.  Please can you close this issue.

dpfaffenbauer, 2016-01-09:
is there any solution yet? i experience the same problem with a turnigy 9x8cv2 receiver.

McJohn115, 2016-01-09:
Sry i didnt find any solution, im using LibrePilot now, works excellent for me and 3D flying is possible.

SteveAmor, 2016-01-09:
Please close this issue.

closed 2016-01-09 by McJohn115