cleanflight/cleanflight #1558: Cleanflight hangs when using MSP on SOFTSERIAL

krzysztofmatula, 2015-12-10:
I'm trying to retrieve telemetry from Cleanflight and pass it to my eLeReS receiver. The receiver supports MultiWii telemetry by using 2 lines RX+TX and sequentially asks MultiWii commands and parses responses for telemetry elements. To have this work UART must be configured as MSP and not "MSP telemetry".

And now the case: it works well when using HW UART. When using SOFTSERIAL instead, Cleanflight is immediately killed, turns on all PCB LEDs and stops responding...

Any suggestions how to pin-point the issue?
I've already asked for the exact sequence the receiver is trying to retrieve from the FC.

I observed identical behavior when using 9600 as well as 19200 bps.
Also, I'm getting the same with 1.10 and 1.11 Cleanflight.

krzysztofmatula, 2015-12-10:
Additional info: I'm using NAZE32 rev5.

Netsplits, 2015-12-17:
I already open a similar issue #1249 it was closed but the problem is there. I think the problem is on softserials, I don't know exactly what is going on, but softserials are not very stable. Check that issue, there is more info there.

deBoulangerie, 2016-01-17:
can you activate softserial in the ports settings with your naze 32 rev 6?
I cant see them.

joshuabardwell, 2016-01-17:
I have the same issue using MSP to change PIDs via MWOSD. I opened an issue here, although I'm a bit too lazy to look it up right now. Just saying, the issue seems to be real.

error414, 2016-01-17:
I have naze32 rev6. And it works me. I connect mw_osd to softserial 1 and i set baudrate to 9600.

joshuabardwell, 2016-01-18:
Mine works fine too. Except every so often the board hangs completely and has to be power cycled.

flyertodd, 2016-03-02:
I have the following setup (Blackbox on UART1, Ublox GPS on UART2, minimOSD micro on Softserial1, HC-06 Bluetooth on Softserial2):
serial 0 129 115200 57600 0 250000
serial 1 2 115200 57600 0 115200
serial 30 1 9600 115200 0 115200
serial 31 0 9600 115200 0 115200

Cleanflight GUI/CLI won't let me turn on MSP on both softserial ports (30,31), hence cannot get both minimOSD & BT to work at the same time. Is this expected?? Both boards are flashed/configured to run at 9600 baud.

joshuabardwell, 2016-03-03:
You cannot have more than two instances of MSP. Assuming you have MSP on UART1 for USB, this is normal behavior.

flyertodd, 2016-03-03:
Oh. Thanks.  Is there any other way to have OSD and BT run on softserial 1 & 2 on the Naze32?  I assume Blackbox Openlog requires MSP, otherwise I could de-select MSP from UART1.

joshuabardwell, 2016-03-03:
Please take this support question over to RCGroups and close this issue on github. I'll be glad to help you over on RCgroups, which is the right forum for support discussions.

flyertodd, 2016-03-06:
Will go ahead and close the issue here.  As temporary relief, I installed a mini slide switch to control the connection of Tx/Rx pins from OSD or BT to softserial 1 (since only one SoftSerial port works, given MSP on UART1).  Later I will hook up a servo to the slide switch for in-flight switching.  Still doesn't solve the problem of not having both softserial ports (OSD & BT) active at the same time.

joshuabardwell, 2016-03-11:
You opened the issue, so you are able to close it by clicking the "close issue" button.

joshuabardwell, 2016-03-11:
PS: Blackbox does not require MSP. It requires Blackbox in the Ports tab. And you can run OSD/MSP and Bluetooth/MSP on Softserial 1 & 2 at the same time, but you have to de-select MSP on UART1 because you can only have two MSP ports running at the same time.

For further discussion, RCgroups is the right place.

flyertodd, 2016-03-11:
Great, I will try replacing MSP from BB UART1 and select it on Softserial2.
Thanks for the tip!

flyertodd, 2016-03-11:
My apologies, but don't seem to be able to close the issue (can't find appropriate button or link). Is it because I did not originally open the issue?

flyertodd, 2016-03-14:
@joshuabardwell - thanks, your suggestion worked.  I have both softserials working w/ MSP, while Blackbox on UART1 is logging data as well.  UART2 has GPS.

The downside is that USB is now disabled on UART1 (since it requires MSP).  So I am forced to communicate with my Flip32+ via BT, which means 1 port will always have to be occupied by the BT module.  Still it serves my purpose for now.

closed 2017-02-25 by hydra